CONTROLLINO-PLC/CONTROLLINO_Library #1: crc-Error at HW-Installation Arduino IDE

sepp68, 2017-02-06:
I get:
[image]

CONTROLLINO-Support, 2017-02-07:
Hello,
I have checked your issue and tried to simulate it on my side. I was able to download and install CONTROLLINO Boards without any problem. 

However, I figured out where your issue might be coming from.
When I downloaded the source files for the boards manually, the downloaded file was corrupted, although the source files on GitHub are alright. If some error during downloading happens to Arduino IDE too, this could be the reason for your error message.

To resolve this issue, please go to folder: "c:\Users\\_username_\AppData\Local\Arduino15\staging\packages\", check the size of your ControllinoHW.zip file and compare it with size of [ControllinoHW.zip file](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/Boards/ControllinoHW.zip) on GitHub, different sizes mean, that the download was unsuccessful and the file should be deleted. Then please, try installing the boards package again.

Here is a screenshot of Checksum Utility tool, comparing SHA-256 checksum in .json file and checksum generated from downloaded ControllinoHW.zip file.
[image]

Here is a screenshot of my Arduino IDE, with successfully installed CONTROLLINO Boards. I am using Arduino IDE version 1.8.1 and Windows 10 Pro 64-bit.
[image]

Best regards,
CONTROLLINO team

sepp68, 2017-02-07:
ok, updating Arduino IDE from 1.6.9 to 1.8.1 did it.
Thank you very much for the quick support

closed 2017-02-07 by sepp68