CONTROLLINO-PLC/CONTROLLINO_Library #4: CONTROLLINO MEGA unknown during compiling

alexmanfrex, 2017-06-30:
Hello,
I'm working on CONTROLLINO MEGA Platform with Arduino 1.8.0 IDE.
I'hadn't any problem in code compiling before this morning.
Despite I follow the right procedure for installing hardware, now, when I try to compile my code, I get this error

"Board controllino_mega (platform avr, package CONTROLLINO_Boards) unknown"

(sorry, I translate it from italian ...)

I can see correctly the list of controllino boards in "Boards Manager" and select them, but at this moment, I can't compile with anything (and so my job is in stand-by ...)

Thanks for your attention

CONTROLLINO-Support, 2017-07-03:
Hello, 
I assume that the problem is cause by one "feature" of Arduino IDE Boards Manager. It does not remove the previous (obsolete) version of the board support package. 
Please look to a folder 

c:\Users\_UserName_\AppData\Local\Arduino15\packages\CONTROLLINO_Boards\hardware\avr\

and delete all obsolete versions.

Please, let us know if it helps.

Thanks,
Best regards,
Lukas

alexmanfrex, 2017-07-03:
Hello,
I fixed this problem on Friday evening and it was actually the problem you describe in your answer …
I’ve removed the folder as you suggest.
Thank for your precious support

Best regards

Alessandro Manfredi
R&D Electronic Designer
Hexagon Manufacturing Intelligence
T: +39 011 4025329  F: +39 011 4025388
E: alessandro.manfredi@hexagon.com

Christian-Thiem, 2017-07-14:
Today I ran into the same issue but on OSX. Where can I find the outdated package to delete it?

CONTROLLINO-Support, 2017-07-17:
Hello Christian, 
please try this path

/Users/_username_/Library/Arduino15/packages/CONTROLLINO_Boards/hardware/avr/

Please, let us know about the results!

Thanks,
Lukas

closed 2017-07-24 by CONTROLLINO-Support