CONTROLLINO-PLC/CONTROLLINO_Library #5: How to get JTAG working with Controllino Maxi

Fuzzyma, 2017-08-02:
I know how to connect my ATmel-Ice to get ISP working on the Controllino. However, connecting to the JTAG Pins with JTAG-Fuse enabled and BOOTRST Fuse disabled is not working.

Is there a way to use JTAG with the Controllino Hardware?

CONTROLLINO-Support, 2017-08-03:
You are correct - due to protection serial resistors on the pin header signals the JTAG does not work. I am afraid that is it not possible to use JTAG with CONTROLLINO.

Fuzzyma, 2017-08-04:
What about debug wire? Is it possible with SPI on Controllino?

CONTROLLINO-Support, 2017-08-07:
I am afraid that there will be the same problem with DebugWIRE as is on Arduino boards - the 100nF capacitor between Atmega16U2 and the main Atmega controller. Please see this article https://awtfy.com/2010/02/21/modify-an-arduino-for-debugwire/ .So, the only way is to disassemble the device and remove the capacitor. Is that a potential way for you?

Fuzzyma, 2017-08-07:
No unfortunately not :/

CONTROLLINO-Support, 2017-08-07:
We are pretty close to release a new generation of CONTROLLINO - https://controllino.biz/controllino-myplc/ It has the JTAG accessible from outside. ;-)

closed 2017-10-26 by Fuzzyma