CONTROLLINO-PLC/CONTROLLINO_Library #7: Allow selection of board library version

dbruneau-ie, 2017-09-14:
Hi,

Is there a way you guys could add the ability for the user to choose the Controllino Library version through the IDE?

I ask because sometimes the codebase for a project could have been done using version 1.1.0 of the library, and using the new version causes issues.

[image]

As an alternative I've tried to force the Arduino IDE to load the library version 1.1.0 by placing it in the sketch folder, but I haven't had any luck in getting that to work.

Regards,

dbruneau-ie, 2017-09-14:
Just wanted to add a note mentionning the following work-around is working for us.

Place the 1.1.0 folder directly in the hardware subfolder of the Arduino IDE.  Make sure to re-organize the folders so they match the image below. (Note there is no folder for the revision number)

Do similar for the libraries.

[image]

You do need to have a copy of the libraries and Board folders already in order to do this.  (Versus having the drop down to choose which version you wish to install directly in the board manager)

CONTROLLINO-Support, 2017-10-12:
Hi,
thank you for your interest in CONTROLLINO! 
The version selection works fine for the library:
[image]

But you are right - for the Boards package it does not work. Thank you that you have found the solution!

We will check how to fix it in the next release of the Boards package.
 
Best regards,
Lukas

per1234, 2018-01-12:
>We will check how to fix it in the next release of the Boards package.

You just need to leave the entries for the previous versions in the JSON file and also leave the files those versions reference available. Looking through the history I can see you did have both entries in a JSON file at one point:
https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/31d16a52a55b0fb2e512b3554160bddee8b746e9/Boards/package_controllinounofficial_index.json
but they both reference the same file name, ControllinoHW.zip. Each version must have a unique file name for this to work.

CONTROLLINO-Support, 2018-01-24:
Thanks for the hint. Will be fixed with the next version.

CONTROLLINO-Support, 2018-08-13:
Fixed by 3.0.1

closed 2018-08-13 by CONTROLLINO-Support