CONTROLLINO-PLC/CONTROLLINO_Library #8: AD Converter reports different values depending on the operating voltage (12V/24V)

Fuzzyma, 2017-10-26:
I connect a voltage source to the A4 Pin on the top of the controllino. The source is at 2V (measured with multimeter).

The controllino voltage reference is defined as INTERNAL2V56.

I power the controllino with USB or 12V (no difference):
- analog read reports ~765

I power the controllino with 24V:
- analog read reports ~735

What just happend? Not only that the behavior is unexpected, it also leads to wrong volatges measured.
Because 765 yields another voltage as 735 (765 beeing correct in that case).

I think this is a big bug. Do you know any solutions to this problem?

CONTROLLINO-Support, 2018-01-24:
I have tested it with the same results.

Please see the [Analog/Digital INPUT block diagram](http://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MAXI-Pinout.pdf). The problem is that Protection resistor and Automatic divider act as a resistor divider. That is the place where "... 0.18V getting lost on the way ...". And the value of the divider changes depending on the supply voltage 12/24V.

Solution is to use screw terminal analog input with defined division ratio: 
12V range: 3,06
24V range: 6,14

msoula, 2018-02-07:
Hi,
The same docs says that there's a serial resistance attached on pin headers x1 and x2. Is there a way to know the resistor value for this resistance ?
FIY, we are planning to connect a NTC 10k probe on these pins but we can figure out how to measure the voltage because of this resistance.
Thanks for your help.

CONTROLLINO-Support, 2018-02-19:
Hi,
pin header in CONTROLLINO is done in following way:
[image]

Where **R1** is:
1 kOhm for: X1.3 – X1.10
220 R for X1.12 to X1.17
1 kOhm for: X2.3 – X2.10
220 R for X2.12 to X2.21
 
Resistor **R2** depends on the input voltage and is:
33 kOhm for 12V
13,24 kOhm for 24V

Is it enough information for you?

CONTROLLINO-Support, 2018-02-23:
We have done a simulation of the circuit and your measured value Vadc of 1.8 V corresponds with NTC resistance of 11.36 kOhm which shall be something between 20-25deg (see the figure below).
In your evaluation of the R4 is calculated value not directly R4, but parallel combination of R4 and R1+R2. Please consider it in your calculation. Please also consider, that NTC is not fully linear.
[image]

msoula, 2018-02-23:
Indeed my calculation was totally wrong. Thanks a lot for your replies.
I think you should add your first pin header diagram into the controllino documentation, it helped a lot.

closed 2020-03-05 by CONTROLLINO-Support

Pulletje, 2020-11-08:
Msoula, Is it possible for you to show your final result. I have the same problem and I can't figure it out

msoula, 2020-11-09:
Sorry @Pulletje, but finally it didn't worked for me. 
I can't remember why it didn't worked exactly...