CONTROLLINO-PLC/CONTROLLINO_Library #11: Error using Controllino libraries in Arduino Web Editor

jm-andonegi, 2018-01-21:
I have tried to use [Arduino Create](https://create.arduino.cc/editor) (web editor of Arduino), with Controllino and I have not been able to compile the examples. I think that the problem is that:

1. Online editor recognises the board as Arduino Mega but it does not allow to load aditional boards as in the IDE.
2. When compiling the program, I get this error:

/home/admin/builder/opt/libraries/latest/controllino-1-1-2/Controllino.h:42:10: error: #error Please, select one of the CONTROLLINO variants in Tools->Board

#error Please, select one of the CONTROLLINO variants in Tools->Board

CONTROLLINO-Support, 2018-01-24:
You are right. Arduino web says "[We are currently supporting only Official Arduino/Genuino boards, and a few Certified and AtHeart boards.](https://create.arduino.cc/projecthub/Arduino_Genuino/getting-started-with-arduino-web-editor-on-various-platforms-4b3e4a)"
Our CONTROLLINO library can be compiled only if you select one of the CONTROLLINO boards.
So, the only way how to use the Arduino Create now (until they add CONTROLLINO to the list of supported devices) is to do not use the CONTROLLINO library and write the sketch like for the UNO or MEGA.

jm-andonegi, 2018-01-24:
A "dirty" trick is to modify the library adding this at the beginning of Arduino.h:

#define CONTROLLINO_MAXI_AUTOMATION

Of course, this is not a solution. It is just hacking the code for a particular board and I just recommend using it in educational situations, where using Arduino Web Editor can be a good solution.

closed 2022-11-18 by pmmarquez