CONTROLLINO-PLC/CONTROLLINO_Library #13: ethernet bootloader

thomaslusli, 2018-05-03:
Tried to use ardiane ethernet bootloader for the Maxi Automation via MEGA 2560 ISP without luck. Is there a ready to use ethernet bootloader?

thomaslusli, 2018-05-07:
Just modified the Ariadne ETH bootloader source code for Maxi Automation. Now loading via tftp is working :)

CONTROLLINO-Support, 2018-05-15:
Cool! May I ask you to share what was the issue with the original Ariadne code, please? Thanks!

thomaslusli, 2018-05-22:
Hi, 

I had no time to modify the toolchain for Controllino. So here are the changes i made.

The original code is:
https://github.com/codebndr/Ariadne-Bootloader

changes to __AVR_Atmega2560__: 
spi.h: 
1.) change Ethernet Interface to PORT J (PORT B is used) and ETH_SS to PINJ3
2.) change Onboard LED to anything you wish (PB7 is used by a DO) or comment in main.c
spi.c:
I have disabled the SD-card.

Make project and use ariadne_atmega2560_w5100

thomaslusli, 2018-06-15:
tftp uploading is working very well until now. I had some troubles with loading code > 32kB. 
The bootloader makes a eth-request after 32 kB to the gateway-address. If there is a unknown gateway in eeprom, the uploading stops and a timeout occurs. A correct gateway solves the problem as described also here: 

https://github.com/codebndr/Ariadne-Bootloader/issues/2#issuecomment-35497441

Sven0909, 2018-08-23:
Hi,
I would like to install the ethernet bootloader on the Controllino MEGA. Should I also change the spi.h/spi.c as @thomaslusli wrote?

Do I really need a programmer or is it possible to use the USB?

Which board I have to choose in the Arduino IDE? The "Atmega 2560" or the old "Controllino MEGA" as I used it for the USB? Especially for installling the ethernet bootloader and after that for normal programming.

Thanks

CONTROLLINO-Support, 2018-08-23:
Hi Sven,

- yes, you have to do the same modification as @thomaslusli. CONTROLLINO MEGA uses the same pins for Ethernet interface as the CONTROLLINO Maxi Automation.

- yes, you really need a programmer. USB programming uses the bootloader you want to re-program :)

- when working with CONTROLLINO, always select proper CONTROLLINO board.

Good luck and let us know about the results!
Lukas

Sven0909, 2018-08-23:
[image]

Thanks for that

Do I select also the proper CONTROLLINO board while I install the ethernet bootloader? The picture says I have to select maybe the "Arduino MEGA 2560  w/ Ariadne Bootloader"?

I will let you know how it works

CONTROLLINO-Support, 2018-08-23:
I would recommend you the [Atmel-ICE](https://www.microchip.com/developmenttools/ProductDetails/atatmel-ice) programmer.

I prefer to do not use it with Arduino IDE, but with Atmel Studio.
Please, see our [guide](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/tree/master/Bootloaders/MEGA) how to program a bootloader to CONTROLLINO MEGA.

redmuchroom, 2019-08-09:
I know it is an old stuff but since it is still open issue :).
I tried to do it (with ariadne bootloader) on a Mega... I did the changes reported by thomaslusli and also some changes in spi.h for the rtc pinout...

I see different issues:
- I was able to upload a sketch on tftp only once... after I cannot ping the mega anymore... this can be solve by writting the ip adress again in the sketch ... however  I  cannot upload by tftp...
- if I reset (reset button), the mega will be remaining in bootloader... it will not launch the sketch...only a power cycle will launch the sketch
- the reset from ethernet doesn't work
- cannot load a sketch by serial from another computer 

i fyou have any suggestion feel free :). 
 if I get some improvements, I will post them.

thanks

thomaslusli, 2019-08-09:
Hallo,

The only point I can agree, is the reset button. I have ignored this function, since the control is remote and I dont need it.

What I can tell you:
- upload is working via tftp every time 
- resetserver is working too, without the server i could not load the sketch remotely
- If I use serial upload (i'm using avrdude) is working too.

For me it sounds, that you have several problems
- Did you write the eprom values used by the bootloader? Than you are not allowed to write on the eprom range of the bootloader values. Did you check the correct values in eprom? 
- Did you start the resetserver in your sketch? Pay attention that you have a free available socket...

Best regards
Thomas

redmuchroom, 2019-08-09:
hi thomas,

thanks for your reply.
I managed to reset it remotely :) thanks for your help... I got the issue on the second point you mentioned :).
so far, I can upload and reset it remotely. I need to figure out the serial part but I think this should be a stupid error again :)

thanks a lot!

hplus28, 2019-11-07:
Hi folks,

I also tried to do the modifications in the spi.h file and compiled the bootloader.
i managed to burn it with an avr isp II and the arduino ide. I am sure the ide is using the modified file.
After this i was unable to upload a sketch using the usb port. 
The hex file is 11 072 bytes long.
Did i miss something? I didn't try to upload via tftp yet, according to ariadne the bootloader is supposed to work with serial too isn't it?

Controllino could provide the tftp bootloader ?

Thank you for your help

Franky

redmuchroom, 2019-11-08:
hi Franky,

Yes you should be able to use the usb port. could you try  to launch the monitor, first? then the controllino should reboot and stay in bootloader stage and you should be able to upload the sketch. (but it is a quite long time i didn't try through usb...)

did you manage to upload the sketch with tftp? 

hope this will help...

seb

hplus28, 2019-11-15:
Hi,
sorry for the delay and thank you for your answer. I didn't try using tftp yet. I am unable to upload via usb.
Here is the modified part of the spi.h file:
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
        /** SPI port and pins */
        #define SPI_DDR         DDRB
        #define SPI_PORT        PORTB
        #define SCK             PINB1  //Pin 52
        #define MISO    PINB3  //Pin 50
        #define MOSI    PINB2  //Pin 51
        #define SS      PINB0  //Pin 53
        /** Ethernet Shield SS pin */
        #define ETH_DDR         DDRB
        #define ETH_PORT        PORTJ
        #define ETH_SS          PINJ3  //Pin 10
        /** SD card SS pin */
        #define SD_DDR  DDRG
        #define SD_PORT PORTG
        #define SD_SS   PING5  //Pin 04
        /** Onboard LED is connected to pin PB7 on Arduino Mega */
        #define LED_DDR     DDRB
        #define LED_PORT    PORTB
        #define LED         PINB6

I didn't change the spi.c file (i didn't understand where to disable the sd card. Maybe this is the problem?

Then i type : "make atmega2560_w5100"
i can seem that the rebuild of the hex file is sucesfull. I then upload this file using arduino ide and the avrisp mkII. The device does not respond anymore after this. I have to reflash the bootlader provided by controllino to make it work again...

Thank you

redmuchroom, 2019-11-19:
hi,
sorry for late reply,  for spi.c &spic.h :check : [(https://github.com/loathingKernel/ariadne-bootloader/files/1939613/Controllino.zip)] (sent be jkroeze on [https://github.com/loathingKernel/ariadne-bootloader/issues/28])

also after the upload of the bootloader do you have aled blinking? (this tells you that the controllino is in boot mode and ready to get a sketch...

hplus28, 2019-11-22:
Hi,
I tried the bbotloader in the zip file and it works, i managed to upload via usb and tftp. Sometimes the controllino seems to get stuck in the bootloarder (the led keeps blinking). I'll check in when this happens. 
Thank you

closed 2020-03-05 by CONTROLLINO-Support