CONTROLLINO-PLC/CONTROLLINO_Library #15: reset programmatically?

andre-pt, 2018-05-07:
is it possible to do a reset programmatically?

thanks

f1oat, 2018-05-08:
Did not manage to find any full software solution for my Controllino MAXI.
So I have added a wire between pin 6 and 11 on X2 connector.
And I use the following code to trig the reset
```
pinMode(45, OUTPUT);
digitalWrite(45, 0);	// X2 connector bridge between pin 6 and 11
```

andre-pt, 2018-05-15:
hi
thanks for you reply, however I have the Pure version without headers.

CONTROLLINO-Support, 2018-05-15:
Enable watchdog, do not service it and wait until it resets the device is not an option?

andre-pt, 2018-05-15:
hmm not sure how to do that, is that documented somewhere?

CONTROLLINO-Support, 2018-05-15:
There is a nice PDF available [here](https://forum.arduino.cc/index.php?topic=63651.0) .

f1oat, 2018-05-15:
I tried with wdt_enable(), and I had to switch to pin 6-11 bridge solution because the Controllino was not fully reset. As far as I can remember, Ethernet was not reset.

CONTROLLINO-Support, 2018-05-17:
You are right, Ethernet chip is not reset by reset wire, but there should be a SW reset done by `Ethernet.begin();` call. This call should lead to call of `W5100Class::init(void)` and it should perform
 `SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
  writeMR(1<<RST);
  writeTMSR(0x55);
  writeRMSR(0x55);
  SPI.endTransaction();`

But this part of the Ethernet library is also not perfect as you can see [here](http://forum.arduino.cc/index.php?topic=148828.15).

What was the behaviour of the Wiznet after the WDG reset, please?

goldberggg, 2018-08-31:
Hi sir. How do I do a reset on controllino mini using arduino software codes. I don't want to manually press the button every time it power up because it still remember the previous error.  Thanks

CONTROLLINO-Support, 2018-08-31:
Use the watchdog as described in this thread. I do not understand the ".. it still remember the previous error ..." part.

goldberggg, 2018-09-12:
Hi sir,

Thank you for your help. 
I have managed to create what I wanted except for the very last part. 

This is the scenario. 
I am using a retro reflective sensor for cross slot wafer detection mapping purpose.

The Elevator will move down. When the sensor sense a wafer present, it will output "1" to the serial reading. When it didn't sense a wafer, it will output "0" to the serial reading.

I connect the sensor amplifier output to pinheader D2 and default baud rate 9600.
The. Program is fresh with just. The declaration of D2 input and serial print. 

Here is the problem. The Elevator takes about 3 secs to move from slot 1 to slot 25.
The sensor can react real time as seen by the on/off at the amplifier. But the serial reading is giving intermittent reading. The arduino serial reading seems like it is lagging and not capturing the amplifier output in real time.
The amplifier is already set to high speed mode. 

Is there anyway to increase the capturing rate at the software?

Thank you!

CONTROLLINO-Support, 2018-09-13:
Hi,

to be honest, I do not understand your description. Maybe you can do some drawing or photo of the setup. And you can also tell us the exact type of the "retro sensor". And maybe you can also share the sketch.
Thanks!

closed 2020-03-05 by CONTROLLINO-Support