CONTROLLINO-PLC/CONTROLLINO_Library #17: installation error Controllino boards

dadoblu2000, 2018-06-25:
I'm trying to install CONTROLLINO but this error is returned how can I solve it?
[image]

CONTROLLINO-Support, 2018-06-25:
Is it the first time you are installing CONTROLLINO? Or is there a possibility that you have some old version of the CONTROLLINO installed already? Thanks!

dadoblu2000, 2018-06-25:
thanks for the quick reply, first time

dadoblu2000, 2018-06-25:
I downloaded the installation guide of the librinomaxi libraries and managed to correctly configure arduino 1.8.5
This topic can be closed thanks

dadoblu2000, 2018-06-25:
I downloaded the installation guide of the controllino libraries and managed to correctly configure arduino 1.8.5
This topic can be closed thanks

CONTROLLINO-Support, 2018-06-25:
Great! Thanks!

CONTROLLINO-Support, 2018-08-13:
Error with downloading of the BSP zip file, or with wrong CRC checksum of the file are time to time reported. But during our testing we have found that it is enough to try to install it e.g. 5 times and then it works. It seems to be an issue between Arduino IDE and GitHub.

closed 2018-08-13 by CONTROLLINO-Support