CONTROLLINO-PLC/CONTROLLINO_Library #18: Controllino Capacitive Library

goldberggg, 2018-07-12:
Hello!
I am trying to use capacitive sensing function in controllino. 
I am able to use it in Arduino with the following codes with serial reading. 
<CapacitiveSensor.h>

CapacitiveSensor cs_4_2 = CapacitiveSensor(4,2);

I have imported the library to controllino but there are no results. Even added resister to both input  A4 and A2 pins. 
<CapacitiveSensor.h>

CapacitiveSensor cs_4_2 = CapacitiveSensor(CONTROLLINO_A4,CONTROLLINO_A2)

Can anyone help with the codes and wiring?
I am stuck for 3 weeks with no progress. 
My intention is to have capacitive function and several sensor input that is why I choose controllino plc. 
Thanks a lot!!

CONTROLLINO-Support, 2018-07-12:
Hello!

Please note that 4 and 2 are different inputs than A4 and A2. 
Maybe you can provide some drawing or photo of your wiring.
Are you using the pinheader (5V input - like Arduino) or the screw terminals (12/24 V input)?  

Which kind of Arduino do you use? There are also some 3,3 V... 
Which type of CONTROLLINO do you have?

Thanks!

goldberggg, 2018-07-12:
Hello!
Thank you for your reply!
I use Arduino UNO, able to get capacitance reading with the codes below and Wiring attached.
When there is object approaching the copper foil, the reading will change which is what i want. Great!
#include <CapacitiveSensor.h>
CapacitiveSensor   cs_4_2 = CapacitiveSensor(4,2);
long total1 =  cs_4_2.capacitiveSensor(30);
 Serial.print(total1);
[image]

I am using CONTROLLINO MINI. I want to get capacitance reading but it always show 1.
Is there any alternative ways to get around this? Can PLC detect capacitive reading as like Arduino did?
#include <CapacitiveSensor.h>
CapacitiveSensor   cs_4_2 = CapacitiveSensor(CONTROLLINO_A4,CONTROLLINO_A2);
long total1 =  cs_4_2.capacitiveSensor(30);
Same connection setup as Arduino.

My project involves getting a capacitive reading when object is nearby and several sensor readings, that is why I choose CONTROLLINO MINI PLC for convenience.

Thanks in advance!

CONTROLLINO-Support, 2018-07-13:
Hello! Thank you for your feedback. I am still missing the information which connector and pins you are using at CONTROLLINO (pinheader or screw terminal? Which pins?)
This kind of sensing will probably not work at 12/24 V screw terminals and may work at the pin header. "May work" means that the protection serial resistors may influence the behaviour.

goldberggg, 2018-07-13:
Hi! I am using Screw input terminal A4 and A2 with no reading.
I see. Can you advise the programming codes on CONTROLLINO MINI for this capacitive sensing function? Do I still need to add the resister if I connect to the pin header?
Thank you!

CONTROLLINO-Support, 2018-07-13:
Hi! Just try to use the same setup and the code like at Arduino UNO and connect it to the MINI X1 pin header. Chose some Digital pins. For the pinout please see [CONTROLLINO MINI](https://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MINI-Pinout.pdf)
Please, let us know about the results!

goldberggg, 2018-07-13:
Hi. thank yo for the fast reply!
Just to clarify. I should connect to one header pin for two header pins?
One of them is sender. One of them is receiver same as Arduino?
Thanks!

goldberggg, 2018-07-17:
Hi sir, I have tried with the pin Head as advised with the same connection as Arduino. There is still no reading. Can advise the codes?
How do. I declare digital pin 5 and 7?
Thanks

goldberggg, 2018-07-17:
I can only get - 2 reading with the digital pins. There is no change in the reading.
It would be great if you can test on your side and see whether it works. Thanks!

CONTROLLINO-Support, 2018-07-17:
There are two simple way how to address CONTROLLINO I/Os

1. The blue bubbles in the [Pinout](https://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MINI-Pinout.pdf) are the corresponding Arduino pins. So Digital5 is 9 and Digital7 is A5.

2. Or you can use [CONTROLINO aliases](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library#aliases-for-the-pins) like CONTROLLINO_D5 and CONTROLLINO_D7.

CONTROLLINO-Support, 2018-07-17:
Regarding the -2 it seems to me that you have some mismatch between used pins and aliases in the code. Please share your code and photo of the test setup. Thanks!

goldberggg, 2018-07-17:
[image]

goldberggg, 2018-07-17:
Hi sir,
This is my code. 

#include <CapacitiveSensor.h>

CapacitiveSensor   cs_5_7 = CapacitiveSensor(5,7);        

void setup()                    
{
   Serial.begin(9600);
}

void loop()                    
{
    long total1 =  cs_5_7.capacitiveSensor(30);

    Serial.print(total1);                  // print sensor output 1
    Serial.print("\t");
}

CONTROLLINO-Support, 2018-07-17:
Please include Controllino library and try this:

`CapacitiveSensor cs_5_7 = CapacitiveSensor(CONTROLLINO_D5,CONTROLLINO_D7);`

goldberggg, 2018-07-18:
Hi sir, tried your codes but still give - 2 result.. Thanks

goldberggg, 2018-07-18:
Hi sir, can you try this simple experiment on your side with controllino mini. I have ran out of option on my side. Thanks

goldberggg, 2018-07-18:
Hi sir, another point worth mentioning is when i Measure the digital pins at arduino board, it gives me 0.4vdc.
However,when I Measure at the pinheader controllino mini, it always give me 0vdc.
Thanks

goldberggg, 2018-07-19:
Hi sir, any update? My project is behind deadline.. Thanks thanks

goldberggg, 2018-07-23:
Hi Sir, sorry to rush you. May I ask is there any update? Thanks

CONTROLLINO-Support, 2018-07-23:
At first, short theory about the CONTROLLINO pins at the pin header

-  D0 - D7 are internally connected to high-side switches with input resistance about 5 kOhm
- A0 - A5 are internally connected to 33 kOhm (for 12V input voltage) 
- Int0 and Int1 are internally connected to 33 kOhm (for 12V input voltage) 

For your high impedance circuit this kind of load means more or less short circuit to ground. That is the reason why you measure 0 V.

The only possible solution we see is to use one of the SPI pins, because these pins have only ESD protection and no additional load. 

Please try following:

- use pin /SS as receive pin (reconnect the wire from D7 to /SS at X1)
- switch OFF the RTC
- update your code in this way

`CapacitiveSensor cs_5_7 = CapacitiveSensor(CONTROLLINO_D5, CONTROLLINO_PIN_HEADER_SS);`

... and let us know about the results!

goldberggg, 2018-07-25:
Hi sir, thank you for the explanation. It works like a charm.
Appreciate your time and effort. 
I will purchase more in the near future. Thank you again!

goldberggg, 2018-08-29:
Hi sir. Controllino PLC D4 start to flicker everytime I connect USB to it. D4 led starts to flicker even faster after connecting to +12 power supply and stop the program thereafter

Does it mean the PLC is faulty?
What should I do to fully reset the PLC? 
I am stuck for days on this.. Kindly advise.. Thanks!

CONTROLLINO-Support, 2018-08-29:
Hello!

Are you able to upload a new sketch, e.g. some simple example? Are there any other wires connected to the device?

Thanks

goldberggg, 2018-08-29:
Hi Sir,

The problem has been solved. It is due to capacitive pinhader. However, I encounter another problem.
I have a 24 DC Motor. There two wires which form a closed loop in normal condition.
It must be open loop in abnormal condition

i tried to use single output D0 but it failed. I was thinking to use R) but it is 240vac.
May I know what codes and output terminal should I use on Controllino MINI?

[image]

goldberggg, 2018-08-30:
Hi sir,
Can I use 240 vac relay CONTROLLINO_R0 for this? Thanks

CONTROLLINO-Support, 2018-08-30:
Hello, you description is not clear to us. Has the motor separate external power supply? 
If you just need to close the loop to run the motor then of course you can use the relay.

closed 2020-03-05 by CONTROLLINO-Support