CONTROLLINO-PLC/CONTROLLINO_Library #19: Analog input Controllino Maxi pure

MugenSAS, 2018-07-26:
Hi everyone,
I want to use an analog sensor which delivers measurments with 4-20 mA current. Is it possible to use an analog input port of a Controllino Maxi pure ? What twill be the value returned by the Controllino when I read it on the analog port ?
Thanks in advance for your answers,
Tristan @ Mugen

frhun, 2018-07-26:
First off sadly you cant get the full 0-1023 range without pretty complex subtracter circuitry - at most 205-1023.
But getting there is quite easy just connect your input to ground with an appropriate resistor.
R = U/20mA
where U is your maximum voltage corresponding to 20mA.
> 12V supply 1023 =^= 15.2V
> 24V supply 1023 =^= 30,4V
Assuming your source device can at most deliver the supply voltage that would be
12V > 600Ω
24V > 1200Ω
with a value range of 162 - 808
( 4mA =^= 162 ; 20mA =^=808 )

Making this a 9.33 bit resolution ;P

MugenSAS, 2018-07-26:
@frhun thanks for your reply. So if I undertand correctly, I have to convert the current ouput from the sensor to a voltage output using a resistor right ? I'm using 24 VDC power supply as it used in an industrial environment so 1.2 kΩ it will be.
> [...] just connect your input to ground with an appropriate resistor
I'm not sure of what you mean exactly, the sensor's output shouldn't be connected to the analog input port of the Controllino ?

MugenSAS, 2018-07-26:
OK understood. How did you determine the range values ( 4mA =^= 162 ; 20mA =^=808 ) ?

MugenSAS, 2018-07-26:
Don't bother, I found out. Thanks again.

closed 2018-07-26 by MugenSAS