CONTROLLINO-PLC/CONTROLLINO_Library #20: Updates needed to support Ethernet version 2.0.0 library

PaulStoffregen, 2018-08-11:
I recently released version 2.0.0 of the Arduino Ethernet library.

The good news is version 2.0.0 greatly improves ethernet performance with the W5100 chip.  It's about 8 times faster (on 8 bit AVR) for TCP to hosts on the same LAN.  Speed to internet connected hosts varies, but it's still a dramatic improvement over prior versions.  More details at these pages:

https://github.com/arduino-libraries/Ethernet/releases/tag/2.0.0
https://www.pjrc.com/arduino-ethernet-library-2-0-0/

The bad news: at least 3 of your customers have reported incompatibility with Controllino.

Ethernet 2.0.0 requires 2 minor updates in your board support package.

1: You need to assign a digital pin number for PJ2.  Really, all that's required is adding an entry in the digital_pin_to_port_PGM[] and digital_pin_to_bit_mask_PGM[] arrays in variants/mega/pins_arduino.h.  If you simply add PJ7 to the end of these 2 arrays, then PJ2 will become digital pin 70.

2: In your pins_arduino.h file, add this line to tell Ethernet 2.0.0 to use pin 70 for the CS signal:

    #define PIN_SPI_SS_ETHERNET_LIB 70

With these changes, Ethernet 2.0.0 should "just work" with your Controllino products.

Or, if you prefer, just copy the latest Ethernet and patch the chip select code in w5100.h, as you have done with prior versions.  However, PIN_SPI_SS_ETHERNET_LIB is meant to give you a long-term stable way to configure which pin the Ethernet library uses for chip select, so you won't need to keep patching the library as we update it and add new features.

CONTROLLINO-Support, 2018-08-12:
Dear Paul,

thank you very much for the hints. For sure we will update the CONTROLLINO board support package!

Lukas

CONTROLLINO-Support, 2018-08-13:
Fixed by BSP 3.0.1

closed 2018-08-13 by CONTROLLINO-Support

PaulStoffregen, 2018-08-13:
Did you define PIN_SPI_SS_ETHERNET_LIB in pins_arduino.h or any other headers?  I don't see it, but maybe I just missed it?

CONTROLLINO-Support, 2018-08-13:
It is done separatelly for each variant, e.g. in variants\Controllino_mega\pins_arduino.h .

PaulStoffregen, 2018-08-13:
Oh, I see now, looks like it's inside the ZIP file.

CONTROLLINO-Support, 2018-08-13:
I have tested it with Ethernet 2.0.0 and it works fine.

CONTROLLINO-Support, 2018-08-13:
May I have one additional questions, please? Do you know if I can have the BSP package without "cores", "libraries", platforms.txt and programmers.txt? I would prefer to use the standard ones for CONTROLLINO boards compilation. But Arduino IDE won't compile without them in ...Local\Arduino15\packages\CONTROLLINO_Boards\hardware\avr\3.0.1\ .

PaulStoffregen, 2018-08-13:
I've not personally used that feature, but I can point you to its documentation.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

Scroll down to "Referencing another core, variant or tool"

PaulStoffregen, 2018-08-13:
FIY - planned for later this year is contributing Teensy's HardwareSerial support for automatic control of RS485 DE to mainstream Arduino.  So a good thing you've added the DE signal as a proper digital pin.  :)

ivan4th, 2018-10-23:
Ok got it working by replacing `pins_arduino.h` under `variants/mega` with one from Arduino IDE completely, along with adding `#define PIN_SPI_SS_ETHERNET_LIB 70` in several places ... The original issue description *contains a mistake*, you can't just add `PJ` to `digital_pin_to_bit_mask_PGM[]`. That's the line in question from Arduino IDE file:
```
_BV( 4 )	, // PJ 4 ** 80 ** CONTROLLINO MEGA Digital 23
```

ivan4th, 2018-10-23:
Seems like `Controllino_HW.zip` isn't handled properly by PlatformIO or something like that

CONTROLLINO-Support, 2018-10-24:
Hi Ivan, thanks for the information! Any idea what should we change in the `Controllino_HW.zip` to be compatible with the PlatformIO? Maybe we can open a separate thread for this issue...

CONTROLLINO-Support, 2018-11-09:
Hi Ivan, we have investigated the issue little bit with following results:

- New installation of PlatformIO includes old version of the Arduino Ethernet library. It must be manually updated to the new version. (remove the old one and install the latest one)
- New installation of PlatformIO includes old version of the CONTROLLINO boards package - again it has to be updated manually.
- PlatformIO has all variants in one folder (as you already mentioned `.platformio/packages/framework-arduinoavr/variants` ) and it seems to be not possible to have different version of file variants/mega/pins_arduino.h for Arduino boards and for CONTROLLINO boards as we have in Arduino IDE. So I am afraid that the only way is what you have already found - replace the original one, by the CONTROLLINO one.

CONTROLLINO-Support, 2018-11-09:
I have added it also to our [FAQ](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/README.md#faq).

sunbit, 2019-01-28:
About the third point, It is no longer needed any patch or replacing any file, i've reached the platformio guys and already integrated that in the latest release. See issue for more information:

https://github.com/platformio/platformio-pkg-framework-arduinoavr/issues/17

Just tested it on my Controllino Mega, with latest platformio 3.6.4, and upgraded framework-arduinoavr to version 1.10623.190125.

Can you update the readme?

Thank you!

PaulStoffregen, 2019-01-29:
@ivankravets - Maybe update the Ethernet library sometime?

> New installation of PlatformIO includes old version of the Arduino Ethernet library. It must be manually updated to the new version. (remove the old one and install the latest one)

ivankravets, 2019-01-29:
I updated all default libraries. Please run `$ pio update`. 

Thanks!

ivankravets, 2019-01-29:
> Update: The Ethernet library 2.0.0 still has to be installed manually :-(

Which version do you see here `$HOME_DIR/.platformio/packages/framework-arduinoavr/libraries/Ethernet/library.properties` ?

CONTROLLINO-Support, 2019-01-29:
We can see there 2.0.0 which is correct, but a new project for CONTROLLINO MAXI, 
 `c_cpp_prc_cpp_properties.json`  includes path:
`$HOME_DIR/.platformio/packages/framework-arduinoavr/libraries/__cores__/controllino/Ethernet/src`
where is: name=Ethernet version=1.1.2 author=Various (see AUTHORS file for details) maintainer=Arduino info@arduino.cc

Frozen copy of old Ethernet library is not a part of our CONTROLLINO Board Support Package from version 3.0.2 released 10 Sep 2018.

So it seems that the PlatformIO inlcludes some old CONTROLLINO BSP. Is that possible, please?

ivankravets, 2019-01-29:
Now I understand, you switched to official Arduino core. So, we've just rolled out a new release:
- https://github.com/platformio/platform-atmelavr/releases/tag/v1.12.3

Please run `pio update`, it should update dev/platform and related packages.

Does it work now?

CONTROLLINO-Support, 2019-01-30:
Thank you for your efforts! Yes, correct, CONTROLLINO BSP does not include any core libraries now. 
We have tried to run `pio update` and also re-instal PlatformIO, but the Ethernet library is completelly missing now. It seems to do not be a part of the standard Arduino core libraries. So, again, it has to be manually installed. To be honest, I am not sure how it is now with Arduino IDE and Ethernet library 2.0.0 - is it part of the core built-in libraries?

Fresh installation of PlatformIO, new project for CONTROLLINO MAXI board and `c_cpp_prc_cpp_properties.json` includes only:
```
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/cores/arduino",
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/variants/Controllino_maxi",
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/EEPROM/src",
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/HID/src",
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SPI/src",
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial/src",
"C:/Users/jirij/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src",
"C:/Users/jirij/.platformio/packages/tool-unity",
"c:/Users/jirij/Documents/PlatformIO/Projects/Controllino test/include",
"c:/Users/jirij/Documents/PlatformIO/Projects/Controllino test/src",

```
There is only one `Ethernet.h` in the `packages`, but some strange version
`c:\Users\jirij\.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\Ethernet\
`
Any idea what is wrong?

Thanks for your support!

ivankravets, 2019-01-30:
Yes, we removed 3rd party libraries from generic Atmel AVR package to avoid similar issues. This is a historical problem of Arduino IDE. It didn't have library manager before, that is why they shipped popular community libraries together with Arduino IDE.

Today, it does not make sense. You can use Arduino IDE library manager and install a library what you need. However, Arduino IDE has a limited library manager, it allows to have the only ONE library with OVE version for ALL your sketches. 

PlatformIO provides professional instruments to manage project dependencies and keep them isolated from other projects. 

Let's back to your issue. You depend on Ethernet library >= 2.0.0 and <3.0.0. So, please explain that PlatformIO and its LIbrary Dependency Finder in `platformio.ini`

```ini
[env:controllino_maxi]
platform = atmelavr
board = controllino_maxi
framework = arduino
lib_deps = Ethernet@^2.0.0
```

See docs https://docs.platformio.org/en/latest/librarymanager/quickstart.html#project-dependencies

CONTROLLINO-Support, 2019-01-30:
OK, clear! Thanks for the explanation and solution! Just one note - Ethernet library requires SPI library and the compiler complaints even if the SPI is already in the built-in libraries. `lib_deps = Ethernet@^2.0.0, SPI` solves the situation. I will update our FAQ in this way.

ivankravets, 2019-01-30:
Great! See you at Embedded World 😊

CONTROLLINO-Support, 2020-04-06:
Hi Ivan, 

I have been informed that the PlatformIO uses some old version of our libary. What should I do to update it? 

Original report from our customer: 

> Could you please update controllino verison in platformio:
https://platformio.org/lib/show/397/CONTROLLINO
The last update is 9 month ago.

Thanks!
Lukas

ivankravets, 2020-04-06:
Hi,

The latest version is 3.0.5 https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/library.properties

CONTROLLINO-Support, 2020-04-07:
Hi, thank you for you fast reply! You are right - the library is up to date. Maybe the customer mixed up the library with the board support package ... So, in general, we do not need to do any action when we release a new version of the BSP or the library? PlatformIO will be updated automatically? Thanks a lot!

ivankravets, 2020-04-07:
We use official Arduino.CC AVR core for Controllino boards. See https://github.com/platformio/platform-atmelavr/blob/develop/boards/controllino_maxi.json#L3

Is this the problem?

CONTROLLINO-Support, 2020-04-07:
I am not sure ... our official JSON is here 
[https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/Boards/package_ControllinoHardware_index.json](url)

ivankravets, 2020-04-07:
What is the difference between Controllino AVR Core and official Arduino?

CONTROLLINO-Support, 2020-08-10:
Our board support package does not include any special Core. We also depend on the official Arduino Core. We just override the `pins_arduino.h` for variants mega and standard.