CONTROLLINO-PLC/CONTROLLINO_Library #22: UPGRADE TO NEW BSP VERSION 3.0.1 AVAILABLE

apetitjean, 2018-09-06:
Hi there,

I'm pretty noob in the Controllino's world and pardon my ignorance but what BSP stands for?

How can determine my current BSP version number and how can I upgrade?

Thanks in advance.

CONTROLLINO-Support, 2018-09-06:
[image]
[image]
[image]

apetitjean, 2018-09-06:
Awesome! Thank you very much. 

And what about the meaning of BSP?

CONTROLLINO-Support, 2018-09-06:
Hi! You are welcome!

BSP means Board Support Package.

Enjoy your CONTROLLINO! :-)

apetitjean, 2018-09-06:
Ah!! This is clearer now! :)

Thanks

closed 2018-09-06 by apetitjean