CONTROLLINO-PLC/CONTROLLINO_Library #23: aREST Library compilation error since update in BSP v3.0.1

apetitjean, 2018-09-07:
Hello everyone,

I'm facing a compilation issue since I updated my Controllino Maxi with BSP v3.0.1.

Previously it worked like a charm except the mapping PIN numbers were shifted (solved in v3.0.1 apparently).

I also updated the Ethernet Library in v2.0.0, ARest Library in 2.7.3 and Arduino IDE in v1.8.6 (previously 1.8.5).

aREST is a nice library to manage Arduino with REST requests.

I don't know what to do. Please help ;-)

Here's the errors:

```
Arduino : 1.8.6 (Windows 7), Carte : "CONTROLLINO MAXI"

G:\LABS\02_Arduino\TP_Ethernet_ARest_v4_controllino.ino\TP_Ethernet_ARest_v4_controllino.ino.ino: In function 'void loop()':

TP_Ethernet_ARest_v4_controllino.ino:60:21: error: no matching function for call to 'aREST::handle(EthernetClient&)'

G:\LABS\02_Arduino\TP_Ethernet_ARest_v4_controllino.ino\TP_Ethernet_ARest_v4_controllino.ino.ino:60:21: note: candidates are:

In file included from G:\LABS\02_Arduino\TP_Ethernet_ARest_v4_controllino.ino\TP_Ethernet_ARest_v4_controllino.ino.ino:16:0:

C:\Users\arnaud\Documents\Arduino\libraries\aREST/aREST.h:734:6: note: void aREST::handle(HardwareSerial&)

 void handle(HardwareSerial& serial){

      ^

C:\Users\arnaud\Documents\Arduino\libraries\aREST/aREST.h:734:6: note:   no known conversion for argument 1 from 'EthernetClient' to 'HardwareSerial&'

C:\Users\arnaud\Documents\Arduino\libraries\aREST/aREST.h:758:6: note: void aREST::handle(char*)

 void handle(char * string) {

      ^

C:\Users\arnaud\Documents\Arduino\libraries\aREST/aREST.h:758:6: note:   no known conversion for argument 1 from 'EthernetClient' to 'char*'

Plusieurs bibliothèque trouvées pour "Ethernet.h"
Utilisé : C:\Users\arnaud\Documents\Arduino\libraries\Ethernet
Non utilisé : C:\Program Files (x86)\Arduino\libraries\Ethernet
exit status 1
no matching function for call to 'aREST::handle(EthernetClient&)'
```

CONTROLLINO-Support, 2018-09-07:
Hi, the problem is compatibility between new Ethernet library and aREST library. The old Ethernet library defines ethernet_h and the new one ethernet_h_ . 
You can easily fix it by one line of code
`#define ethernet_h`
Just place it above line 
`#include <aREST.h>`

To solve this issue properly, author of aREST should update the library header to use also ethernet_h_ .

apetitjean, 2018-09-07:
Awesome! Thank you. I'm gonna test it and go back to you!!! :)