CONTROLLINO-PLC/CONTROLLINO_Library #24: Unix Time stamp?

hl68fx, 2018-09-12:
Hi everyone,

is it possible to get the unix time stamp from the built in realtime clock? It would be cool for countdowns, etc.

CONTROLLINO-Support, 2018-09-13:
Hi,

do you mean "real UNIX time stamp", or just "UNIX like time stamp"? It is not so simple - please go through [this thread](https://forum.arduino.cc/index.php?topic=55433.0).

harbulot, 2018-09-15:
I would also be interested in this, with a millisecond degree of precision. Ideally, I'd like to read a 64-bit integer that represents the number of milliseconds since 1st Jan 1970 00:00 UTC (similar to what you'd get with Java or JavaScript's `Date.getTime()`), ignoring leap seconds for simplicity.
Is this possible?

closed 2022-11-18 by pmmarquez