CONTROLLINO-PLC/CONTROLLINO_Library #25: input pins

lballaty, 2018-09-30:
Hi I need to connnect the controllino input pins to a simple switch - it is a spring loaded switch that makes contact when you press it but then opens again and opens the contact.

I just need to detect logical 1 or 0 on the input - the following pins -

82 -97 and 50,52,70

do I need external resistors to make this work reliably? I'd like to avoid putting another board.

if this is not the right place to ask this question please point me to the right place. I have read the documents and the pinout related drawings but I can't tell if you have something built into the board to allow me to work without additional external resistors

thanks

searchingFool :)

KriMeSoft, 2018-09-30:
I would love to see an actual schematic of the inputs!

CONTROLLINO-Support, 2018-09-30:
Hi! As you can see on the [pinout drawing](https://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MEGA-Pinout.pdf) each Analog/Digital input has an "automatic divider". Part of the divider is a pull-down resistor. So, when the contact is open, there is always logical 0 on the pin. Connect the supply voltage of the CONTROLLINO to the switch to ensure that there will be logical 1 when the switch is pressed. There is no need for additional resistors. 
Good luck! :)

CONTROLLINO-Support, 2018-09-30:
Block diagram of the inputs is on the [pinout drawing](https://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MEGA-Pinout.pdf) and you can find some more information [in this thread](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/issues/8). Do you need more information?

KriMeSoft, 2018-09-30:
A block diagram isn't really helpful. I have seen the other thread, but that doesn't seem to match the block diagram.
The block diagram shows an "automatic divider" applied to the screw terminal only and not the pin header but in the thread it looks as if this divider is also applied to the pin header.
I measure 33kOhm on the pin header but about 100kOhm on the screw terminal, doesn't seem to match what is stated in the other thread.

These are just 5 things of the top of my hat:
- can we drive analog signal 0-5V on the pin header (corresponding to A0-A9) or does it follow the screw terminals voltage levels?
- can we have the divider work with the 12V values even when using 24V power supply, as the 0 to 26,4V input range is not really useful. I've never seen anything else than 0-10V source except for some specialty stuff. (and the automation variant doesn't suit us)
- if we need (extra) 5V outputs, can we use the 'inputs' as outputs through the pin header?
- if we need (extra) 5V inputs, can we use the 'digital outputs' as inputs through the pin header?
- what input circuitry is on TX1 (18) and RX1 (19)?

Can you not supply the actual schematic for the inputs so we can determine all safe and correct ways of using the pin headers and screw terminals?
I doubt there is anything preventing this from being shared?

lballaty, 2018-09-30:
Hi, that is good news …  just to confirm you mean by supply voltage the 12V not the 5V which is next to the RS485 pin … I’m running the controllino using a 12V power supply

thank you

Libor Ballaty
liborballaty@gmail.com

CONTROLLINO-Support, 2018-10-01:
Yes, correct, I mean 12 V.

CONTROLLINO-Support, 2018-10-01:
@KriMeSoft - please contact us directly at info@controllino.biz . Thanks!

closed 2020-03-05 by CONTROLLINO-Support