CONTROLLINO-PLC/CONTROLLINO_Library #26: Pull-up for digital inputs in MAXI Automation results in low level

sumpfralle, 2018-10-22:
I am using a Controllino MAXI Automation.
I tried to use digital input pins with a configured pull-up resistor:
```c
pinMode(SOME_PIN, INPUT_PULLUP);
```

But I am surprised that the pin headers of such digital inputs reach only a level of 1.3V (instead of being close to 5V). Thus I am forced to use analog inputs (with configured pull-up) and use a threshold of ca. 280 for distinguishing HIGH and LOW.

In contrast, the same setup works as expected (with a good HIGH level) with a Controllino Mini.

Is there a technical problem with the MAXI Automation devices with regard to pull-up resistors?
Or do you have an idea, what could have gone wrong with my device?

CONTROLLINO-Support, 2018-10-22:
Hello, the reason is the pull-down resistor. Please, go through [this thread](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/issues/25) to get more information about CONTROLLINO inputs.

sumpfralle, 2018-10-23:
Thank you for your quick response!

Regarding #25: yes, after reading this issue again, it indeed seems to be related.

But since this was not clear in #25: is it really impossible to use `INPUT_PULLUP` as a `pinMode` in a MAXI Automation controller? If yes: is this only valid for this model?
In this case I would have really appreciated a note in the pinout block diagram. The detail diagram with the "automatic divider" (sadly hiding its internal pull-down resistor) could be a good place.

sumpfralle, 2019-02-18:
Ping?

CONTROLLINO-Support, 2019-02-18:
Hello, I am sorry, but I have somehow missed your question. 

Usage of internal pull-up together with external pull-down will result to a voltage divider at the input. This does not make sense from my point of view.

All variants of CONTROLLINO have similar circuitry at Analog/Digital inputs.

My colleague is working on the new pinout block diagrams. I will try to update this thread when they are ready.

sumpfralle, 2019-02-18:
> Usage of internal pull-up together with external pull-down will result to a voltage divider at the input.

Maybe my wording was confusing; I did not use any external resistors in this case. I measured the 1.3V with `INPUT_PULLUP` while that header pin was not connected.

> All variants of CONTROLLINO have similar circuitry at Analog/Digital inputs.

I assumed this. This made the difference between MINI and MAXI even more confusing to me.

dannyhenderson, 2019-03-19:
I have this exact same question. I have a prototype that I built with a standard Arduino and I'm trying to adapt it to Controllino. The devices I am interfacing with signal LOW, so unifying the supply ground for my Arduino and the device allows me to use INPUT_PULLUP on the input pins to use Arduino's internal pullup resistors. Then I simply read the LOW signal directly. If what I'm hearing is correct the Controllino has an internal PULLDOWN resistor, that prevents the use of INPUT_PULLUP which is unfortunate because that would then require an additional external relay to basically change the LOW signals into HIGH signals. Am I missing something?

CONTROLLINO-Support, 2019-03-20:
Hello,
Yes, CONTROLLINO has on input internal pull-down. When you wan to use device on inputs, which needs the external pull-up (I expect some open collector output on the external device), you have to use some more external components to adapt it for CONTROLLINO input. For example logic gate with pull-up on his input side etc. All depends on your application. Input impedance of the screw terminal CONTROLLINO input is approx. 80kOhm so when you will use external pull-up of 8K2 it can also work (approx. division 1:10). For the pinheader you are facing of the pulldown of 15k so 1k5 resistance can work. Of course for this solution you shall be careful about higher currents and power losses (low impedance).

sumpfralle, 2019-03-27:
Thank you for your response.

Just for clarification: this situation is probably not by design, but by mistake, or?
(mistakes are no problem, as long as they are communicated clearly in order to prevent users from wasting their time)

CONTROLLINO-Support, 2019-03-27:
You are welcome! We do not see it as a mistake. Maybe you can advise us how to design a simple voltage divider to measure 12/24V signals and then also be able to pull-up and pull-down?

sumpfralle, 2019-03-28:
I see. Thank you for the explanation.

I created a [pull request](#32) for adding this information to the FAQ.
Maybe you are aware of other places, where this crucial bit of information could be helpful for users?

CONTROLLINO-Support, 2019-03-29:
Thank you for the FAQ! I will inform the team about your suggestion.

closed 2019-03-29 by CONTROLLINO-Support

dannyhenderson, 2019-03-31:
> You are welcome! We do not see it as a mistake. Maybe you can advise us how to design a simple voltage divider to measure 12/24V signals and then also be able to pull-up and pull-down?

If this is not a mistake, would you not agree that "100% Arduino Compatibility" might be a little misleading? It's not to difficult to work around, but the idea that you cannot take an Arduino prototype and run it on Controllino seems to indicate it is not 100% compatible.

dannyhenderson, 2019-05-08:
I see this is marked as closed but you still claim 100% Arduino compatibility on your website. This is not true. It's not a large issue but people should be aware that sketches that work with an Arduino may require external pull-up resistors. I love the product, but please be clear.

troybourne, 2019-12-02:
I agree with this... I was a bit frustrated to find out that such a commonly used function as INPUT_PULLUP is NOT supported after ordering 2 Controllinos.

Jarod997, 2021-02-24:
I have to agree - I had just tried a fairly simple program for a prototype cable tester I'm designing and was puzzled why pinMode(somePin, INPUT_PULLUP); wasn't working as expected.

NOT 100% Arduino compatible.

To be fair, you could implement a hardware switch (eg dip-switch) which would bypass the input divider and make the input function as expected. A prominent note explaining this function should also be included.

PR #32 (2019-03-29): FAQ: describe usage of pull-up mode for inputs
CONTROLLINO boards do not support Arduino's `INPUT_PULLUP` mode, due to their integrated pull-down resistors.

Closes: #26