CONTROLLINO-PLC/CONTROLLINO_Library #28: Board Library

gators54, 2018-11-28:
[image]

I am having trouble getting the board library to load. Everything else is installed properly.

Thanks.

gators54, 2018-12-03:
Yes i did and it worked. This is the link I am using.

https://raw.githubusercontent.com/CONTROLLINO-PLC/CONTROLLINO_Library/master/Boards/package_ControllinoHardware_index.json

gators54, 2018-12-03:
[image]

CONTROLLINO-Support, 2018-12-04:
Then something seems to wrong at your side. Because you should see this:
[image]

gators54, 2018-12-04:
[image]

Sorry, that was the file from the preference folder showing. I get the same thing you do in the browser. I've tried a few different things and still no luck.

CONTROLLINO-Support, 2020-03-05:
Hello, another user has reported the same issue and finally the root cause was some additional space in the URL, or maybe before or after the URL.

closed 2022-11-18 by pmmarquez