CONTROLLINO-PLC/CONTROLLINO_Library #29: ISR USART3_RX_vect Error: multiple definition of `__vector_54'

theblond, 2018-12-12:
Hallo,

i have programmed a DMX reciever on a Controllino MEGA over the RS 485 interface. The compiler crashes with the error: multiple definition of `__vector_54'

It compiles, when I comment out the ISR in my ino-file.
`ISR (USART3_RX_vect) {//Do something}`

Or it compiles, when I leave my ISR as it is and comment out the initialization routine of the RS 485 in the Controllino.cpp and Controllino.h.
`char Controllino_RS485Init( long aBaudrate )`

I hope there will be a clever adaption, so I don't have to modify the library manually.

CONTROLLINO-Support, 2018-12-12:
Hello, 

let´s try to use Controllino_RS485Init( void ). This function just configures the bus direction control signals and does not call Serial3 at all.

theblond, 2018-12-12:
Of course, I can't write my own code for USART3 and use the Arduino Serial3 at the same time.

Sorry I have not been specific at this point, but the curious thing is, I never call Controllino_RS485Init( long aBaudrate ) in my project.

The error already turns up with this few lines of code:

> #include <Controllino.h>
> 
> void setup() {
>   // put your setup code here, to run once:
> 
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
> 
> }
> 
> ISR (USART3_RX_vect) {
>   // do something
> }

The compile error:

> HardwareSerial3.cpp.o (symbol from plugin): In function `Serial3':
> 
> (.text+0x0): multiple definition of `__vector_54'
> 
> sketch\Controllino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
> 
> c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
> 
> collect2.exe: error: ld returned 1 exit status
> 
> exit status 1
> Fehler beim Kompilieren für das Board CONTROLLINO MEGA.

CONTROLLINO-Support, 2018-12-12:
Yes, you are right, the compiler builds also the Serial3 code even if you do not call Controllino_RS485Init( long aBaudrate ). Pretty sad ... I am afraid that you have to comment out the line Serial3.begin(aBaudrate); in our Controllino.cpp. Then the compilation of your example code works fine - as you have already found. Maybe we can add some define in future versions of the library. Like CONTROLLINO_DO_NOT_USE_SERIAL3 :)

CONTROLLINO-Support, 2019-06-20:
Unfortunately, Arduino has troubles to pass defines to the library sources, so my proposal with the CONTROLLINO_DO_NOT_USE_SERIAL3 will be not implemented.

CONTROLLINO-Support, 2019-06-20:
See [this thread](https://github.com/arduino/Arduino/issues/2015) for more details.

closed 2020-03-05 by CONTROLLINO-Support

vallemar, 2021-10-31:
something to fix this? I have come across this problem and I don't know how to solve it

CONTROLLINO-Support, 2021-11-03:
As I have proposed in my previous comment: "...I am afraid that you have to comment out the line Serial3.begin(aBaudrate); in our Controllino.cpp".