CONTROLLINO-PLC/CONTROLLINO_Library #30: Wrong pin ID in for screw terminals of MINI's relays

sumpfralle, 2019-02-18:
I am using a MINI device.

In order to use two of the relays, I had to overwrite two defines:
```
#define CONTROLLINO_SCREW_TERMINAL_RELAY_00 4                                                       
#define CONTROLLINO_SCREW_TERMINAL_RELAY_02 6                                                       
```

The original values seem to be:
```
#define CONTROLLINO_SCREW_TERMINAL_RELAY_00 8
#define CONTROLLINO_SCREW_TERMINAL_RELAY_02 4
```

Or maybe I am just misinterpreting the pinout or the labeling?
I would be glad, if you could verify this.

CONTROLLINO-Support, 2019-02-18:
Oops! You are right! I am going to fix ...

Thanks for the report!

CONTROLLINO-Support, 2019-02-18:
Fixed. Thread to be closed when the new version of the library will be released.

sumpfralle, 2019-06-11:
ping?

CONTROLLINO-Support, 2019-06-13:
We were still waiting for some other stuff to do not release a new version for each small bugfix. But you are right, time is over :)

CONTROLLINO-Support, 2019-06-20:
[Done! Thanks!](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/releases)

closed 2019-06-20 by CONTROLLINO-Support