CONTROLLINO-PLC/CONTROLLINO_Library #31: 10. There is not possible to upload a new sketch to CONTROLLINO anymore

Keiser615, 2019-03-27:
Please help me I have purchased a Controllino MEGA and it was workig well until the issue of the upload ocurred is the anybody who has fixed this Problem??? Please send me a tutorial and the links to download all the files needed to fix it.

direccion.operaciones@automationsoft.mx

CONTROLLINO-Support, 2019-03-28:
Please try the procedure described in our [FAQ](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library#faq) No#10.

Keiser615, 2019-03-28:
We did it just like the FAQ said but nothing happend.

CONTROLLINO-Support, 2019-03-28:
Please enable the verbose output for upload in Arduino IDE Preferences and share with us what avrdude says. 
Do you have something connected to the pin header, please?
Is the Reset LED permanently shining?

Keiser615, 2019-04-02:
This is the sketch the team tried to upload:

#include<SoftwareSerial.h>//Librería serial
#include<Servo.h>//Librería servo
#define outputA 6
#define outputB 7
Servo servo;
int counter = 0; 
int aState;
int aLastState;  
int pos = 0; //Declaración de variables
int pinBluetoothTX=10;
int pinBluetoothRX=11;
int pinServo=9;
int led4=4;
int led1=5;
char letra;
int cont=0;
int start_cycle=0;
int pos_home=0;
int distancia=0;
SoftwareSerial bluetooth(pinBluetoothTX, pinBluetoothRX);

void setup() { //Configuración I/O
  
    Serial.begin(9600);
    servo.attach(pinServo);
    servo.write(90);
    bluetooth.begin(9600);
    pinMode(led4, OUTPUT);
    pinMode(led1, OUTPUT);
    pinMode (outputA,INPUT);
    pinMode (outputB,INPUT);
    aLastState = digitalRead(outputA);}

void loop(){ //Programas a ejecutar
  
    aState = digitalRead(outputA); // Programa encoder
    if (aState != aLastState){     
    if (digitalRead(outputB) != aState) { 
    counter ++;}
    else {
    counter --;}
    Serial.print("Mm: ");
    Serial.println(counter);} 
    aLastState = aState;

    if(bluetooth.available() > 0){
    letra=bluetooth.read();
    Serial.println(letra);
    start_cycle=1;}

    if(letra == '1'){ // Programa para modificar variables
    distancia=100;}
    if(letra == '2'){
    distancia=200;}
    if(letra == '3'){
    distancia=300;}
    if(letra == '4'){
    distancia=400;}
    if(letra == '5'){
    distancia=500;}

    if(start_cycle == 1 && counter >= distancia){ // Programa automático
    servo.write(90);
    digitalWrite(led4, HIGH);
    digitalWrite(led1, LOW);
    delay(2000);
    digitalWrite(led1, HIGH);
    digitalWrite(led4, LOW);
    delay(2000);
    digitalWrite(led1, LOW);
    digitalWrite(led4, LOW);
    servo.write(20);
    start_cycle=999;
    pos_home=1;}
    
    else if(start_cycle == 1 && counter < distancia){ // Programa para mandar a "home"
    servo.write(180);}
    if(pos_home == 1 && counter <=0){
    servo.write(90);
    pos_home=999;}}

Keiser615, 2019-04-03:
We are going to use the bootloader the led blinking is ok but we are going
to need your help after reseting controllino cause we are goint to connect
a Bluetooth device.

Please post us your comments.

CONTROLLINO-Support, 2019-04-03:
Hello, I do not understand the point "we are going to use the bootloader".

If the CONTROLLINO is "bricked" then it does not make sense which sketch do you want to upload. 

Again - Please enable the verbose output for upload in Arduino IDE Preferences and share with us what avrdude says.
Do you have something connected to the pin header, please? Especially to the TX and RX lines ...

Keiser615, 2019-04-04:
Good Afternoon this is the error coment:

Arduino:1.8.8 (Windows 10), Tarjeta:"CONTROLLINO MEGA"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine
-hardware C:\Program Files (x86)\Arduino\hardware -hardware
C:\Users\rdiaz\AppData\Local\Arduino15\packages -tools C:\Program Files
(x86)\Arduino\tools-builder -tools C:\Program Files
(x86)\Arduino\hardware\tools\avr -tools
C:\Users\rdiaz\AppData\Local\Arduino15\packages -built-in-libraries
C:\Program Files (x86)\Arduino\libraries -libraries
C:\Users\rdiaz\OneDrive\Documentos\00_Trabajo\00 Documentos\04 C󤩧o
arduino\Controllino\libraries -fqbn=CONTROLLINO_Boards:avr:controllino_mega
-vid-pid=0X2341_0X0042 -ide-version=10808 -build-path
C:\Users\rdiaz\AppData\Local\Temp\arduino_build_566117 -warnings=none
-build-cache C:\Users\rdiaz\AppData\Local\Temp\arduino_cache_361187
-prefs=build.warn_data_percentage=75
-prefs=runtime.tools.avr-gcc.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.avr-gcc-4.9.2-atmel3.5.3-arduino2.path=C:\Users\rdiaz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2
-prefs=runtime.tools.avrdude.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.avrdude-6.3.0-arduino8.path=C:\Users\rdiaz\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino8
-prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program
Files (x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.arduinoOTA.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr -verbose
C:\Users\rdiaz\OneDrive\Documentos\00_Trabajo\00 Documentos\04 C󤩧o
arduino\Controllino\libraries\CONTROLLINO\Common\Blinking_LEDs\Blinking_LEDs.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine
-hardware C:\Program Files (x86)\Arduino\hardware -hardware
C:\Users\rdiaz\AppData\Local\Arduino15\packages -tools C:\Program Files
(x86)\Arduino\tools-builder -tools C:\Program Files
(x86)\Arduino\hardware\tools\avr -tools
C:\Users\rdiaz\AppData\Local\Arduino15\packages -built-in-libraries
C:\Program Files (x86)\Arduino\libraries -libraries
C:\Users\rdiaz\OneDrive\Documentos\00_Trabajo\00 Documentos\04 C󤩧o
arduino\Controllino\libraries -fqbn=CONTROLLINO_Boards:avr:controllino_mega
-vid-pid=0X2341_0X0042 -ide-version=10808 -build-path
C:\Users\rdiaz\AppData\Local\Temp\arduino_build_566117 -warnings=none
-build-cache C:\Users\rdiaz\AppData\Local\Temp\arduino_cache_361187
-prefs=build.warn_data_percentage=75
-prefs=runtime.tools.avr-gcc.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.avr-gcc-4.9.2-atmel3.5.3-arduino2.path=C:\Users\rdiaz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2
-prefs=runtime.tools.avrdude.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.avrdude-6.3.0-arduino8.path=C:\Users\rdiaz\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino8
-prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program
Files (x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.arduinoOTA.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr
-prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files
(x86)\Arduino\hardware\tools\avr -verbose
C:\Users\rdiaz\OneDrive\Documentos\00_Trabajo\00 Documentos\04 C󤩧o
arduino\Controllino\libraries\CONTROLLINO\Common\Blinking_LEDs\Blinking_LEDs.ino
Using board 'controllino_mega' from platform in folder:
C:\Users\rdiaz\AppData\Local\Arduino15\packages\CONTROLLINO_Boards\hardware\avr\3.0.2
Using core 'arduino' from platform in folder: C:\Program Files
(x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g
-Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections
-fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x
[... 130 lines cut ...]
"C:\\Users\\rdiaz\\AppData\\Local\\Temp\\arduino_build_566117/Blinking_LEDs.ino.elf"
"C:\\Users\\rdiaz\\AppData\\Local\\Temp\\arduino_build_566117/Blinking_LEDs.ino.eep"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O
ihex -R .eeprom
"C:\\Users\\rdiaz\\AppData\\Local\\Temp\\arduino_build_566117/Blinking_LEDs.ino.elf"
"C:\\Users\\rdiaz\\AppData\\Local\\Temp\\arduino_build_566117/Blinking_LEDs.ino.hex"
Usando librer�CONTROLLINO con versi󮠳.0.4 en la carpeta:
C:\Users\rdiaz\OneDrive\Documentos\00_Trabajo\00 Documentos\04 C󤩧o
arduino\Controllino\libraries\CONTROLLINO
Usando librer�SPI con versi󮠱.0 en la carpeta: C:\Program Files
(x86)\Arduino\hardware\arduino\avr\libraries\SPI
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A
"C:\\Users\\rdiaz\\AppData\\Local\\Temp\\arduino_build_566117/Blinking_LEDs.ino.elf"
El Sketch usa 2916 bytes (1%) del espacio de almacenamiento de programa. El
mḩmo es 253952 bytes.
Las variables Globales usan 341 bytes (4%) de la memoria din᭩ca, dejando
7851 bytes para las variables locales. El mḩmo es 8192 bytes.
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
Ha ocurrido un error mientras se enviaba el sketch

techscout1, 2019-10-15:
Hey Guys,
we've the same issue with our Controllino Maxi. We're going to use them as a PLC for the commercial product, but there no possiblity to upload a sketch. 

Everytime we try it, we get this error message:
avrdude: stk500v2_ReceiveMessage(): timeout

The wiring is correct, USB + 12 V. We tried three different PCs (win10/win7), I'm an engineer for embedded systems and have no idea what could be the reason for this fault. Could there be a hardware problem? 

If we can't solve the problem in the next days, we need to switch to an other product for our production. Hoppefully somebody has an idea ;)

CONTROLLINO-Support, 2019-10-15:
Hello, 
just a first fast feedback - have you tried the procedure described in our [FAQ](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library#faq) No#10.

Do you have something connected to the pinheaders?

Thanks!

Keiser615, 2019-10-15:
Maybe you burned the USART 0 port you must use the USART port 1 to program
the Controllino with Atmel Studio we did this procedure and the Controllino
Upload the sketch.

closed 2022-11-18 by pmmarquez