CONTROLLINO-PLC/CONTROLLINO_Library #35: avrdude: verification error

ykSneerG, 2019-12-19:
Hello,

maybe this issue is not related to the Controllino itself. 
Since yesterday I have one Controllino Mini for testing and I get an error while uploading the sketch. After checking the internet I found several reports about this bug and I also tried the FAQ 10 to solve this issue, but without success.

Here is my sketch, that produces the error:
`#include <Controllino.h> 

int t1 = 100;
int t2 = 2000;
int t3 = 5000;

void setup() {
  
  pinMode(CONTROLLINO_D0, OUTPUT);
  pinMode(CONTROLLINO_D3, OUTPUT);
}

void loop() {

  digitalWrite(CONTROLLINO_D0, HIGH);
  delay(2000);            
  digitalWrite(CONTROLLINO_D0, LOW);
  delay(5000);   

  digitalWrite(CONTROLLINO_D3, HIGH);  
  delay(100);                       
  digitalWrite(CONTROLLINO_D3, LOW);    
  delay(2000);              
           
}

For me, it looks that the delay is not handled correctly for whatever reason. If I use the same time for all the delays, it is working.

I tried one further solution and this is also working, but with the same delay as in the non working sketch from above.

`#include <Controllino.h> 

int t1 = 100;
int t2 = 2000;
int t3 = 5000;

void setup() {
  
  pinMode(CONTROLLINO_D0, OUTPUT);
  pinMode(CONTROLLINO_D3, OUTPUT);
}

void loop() {

  digitalWrite(CONTROLLINO_D0, HIGH);
  delay(t2); 
  digitalWrite(CONTROLLINO_D0, LOW); 
  delay(t3);  
  digitalWrite(CONTROLLINO_D3, HIGH);  
  delay(t1);                       
  digitalWrite(CONTROLLINO_D3, LOW);    
  delay(t2);              
           
}`

Here I defined variables outside the loop and than it is compiling and uploading without an issue.

Before I used an Arduino nano without a problem, so this was new for me. What kind of software do I have install, right now I am using the latest Arduino IDE and the suggested setup in the manual. Or is this the correct way to handle the delay settings. Do you need more informations, maybe this issue entry is connected to #31, not sure. Do you need further info?

CONTROLLINO-Support, 2019-12-20:
Hello, thank you very much for the report. It sounds really interesting - maybe it will help us to deal with the problem of bricked Arduino and CONTROLLING devices.

CONTROLLINO-Support, 2020-02-28:
I have some news regarding this topic - we have found some potential root cause of these troubles in the original Arduino bootloader.
Please see [THIS TUTORIAL](https://forum.controllino.biz/viewtopic.php?f=6&t=2199) how to update to the new MINI bootloader.

closed 2020-03-05 by CONTROLLINO-Support