CONTROLLINO-PLC/CONTROLLINO_Library #36: Modbus RTU library

smartynov, 2020-01-09:
I would like to implement a control over several Modbus RTU modules in Controllino. 

This repository gives some examples, but they are somewhat outdated and even dangerous. For example, the following code found in [ModbusRtu.h](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/MEGA/DemoModbusRTUMaster/ModbusRtu.h) 

```
        while ( port->available() )
        {
            au8Buffer[ u8BufferSize ] = port->read();
            u8BufferSize ++;

            if (u8BufferSize >= MAX_BUFFER) bBuffOverflow = true;
        }
```

...doesn't stop on buffer overflow and may just spoil memory beyond 64-byte au8Buffer occasionally. I believe such mistakes should never occur in a code for an industrial-grade PLC.

I found a few other libraries, however they will not work directly because of the non-standard pins.

Can you recommend any Modbus RTU library that works well with Controllino? 

If there is none, probably it is a good idea to write one and include for all Controllino users. I'm ready to contribute to the code.

CONTROLLINO-Support, 2020-08-10:
Hello, thank you for the note about the potentialy dangerous code. But we have just taken some existing Modbus RTU library and prepared an example how to use it. The main reason is just to demonstrate that the CONTROLLINO is ready for the Modbus RTU from the hardware point of view. For sure it is not a solution to be directly used for some industrial application. 
Do you have any progress here, please? Have you found a better library you can recommend to other CONTROLLINO users? Thanks!

paolocavagnolo, 2020-11-14:
Hi,
first time working with MODBUS RTU, so please be patient.

I finally succeed in using a Controllino MEGA as SLAVE, i'm checking it with QModMaster with a USB-RS485 interface.

After trying to implement the Controllino_RS485_pins on different libraries I ended up with [this one](https://code.google.com/archive/p/arduino-modbus-slave/downloads).

The change I've made are all in the modbusSlave.cpp, [modbusSlave.cpp.zip](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/files/5541114/modbusSlave.cpp.zip):

1. add #include <Controllino.h>
2. change _Serial_ in _Serial3_
2. set the _receiver mode_
`digitalWrite(CONTROLLINO_RS485_nRE, LOW);
  digitalWrite(CONTROLLINO_RS485_DE, LOW);`
in two point: in **checkSerial** and of course in **serialRx**
3. set the _transmission mode_
`digitalWrite(CONTROLLINO_RS485_nRE, HIGH);
  digitalWrite(CONTROLLINO_RS485_DE, HIGH);`
in **run**, and also a Serial3.flush after the transmission

The working arduino code is the following:
`#include <Controllino.h> 

#include <modbus.h>
#include <modbusDevice.h>
#include <modbusRegBank.h>
#include <modbusSlave.h>

modbusDevice regBank;
modbusSlave slave;

void setup() {
  regBank.setId(1);
 
 pinMode(CONTROLLINO_RS485_TX, OUTPUT);
 pinMode(CONTROLLINO_RS485_RX, INPUT);
 
 pinMode(CONTROLLINO_RS485_nRE, OUTPUT);
 pinMode(CONTROLLINO_RS485_DE, OUTPUT);

  regBank.add(1);
  regBank.add(2);
  regBank.add(3);

  slave._device = &regBank;
  slave.setBaud(19200); 
}

void loop() {

  regBank.set(1, digitalRead(CONTROLLINO_A10));
  regBank.set(2, digitalRead(CONTROLLINO_A11));  

  digitalWrite(CONTROLLINO_D0,regBank.get(3));
  
  slave.run();  

}`

Tested in both read coil and write coil.

CONTROLLINO-Support, 2020-11-18:
Hi Paolo,

well done! 👍 

Thank you very much for sharing this! 👏 

Lukas

MosiMiso, 2020-12-04:
Hi Controllino Team,
I am going to use a Controllino MAXI as a Modbus Master to read in a sensor.
Because you wrote the library is not intended to use for industrial solutions, i am wondering if you can provide a library for Modbus RTU Master? 
The soution Paolo mentened is for Modbus Slave as far as i saw.

Would be great, if you can provide a working Modbus library for master and slave for Controllino.
Michael

GregDavid123, 2021-10-20:
Hi @CONTROLLINO-Support  @MosiMiso  @paolocavagnolo  ,

What do you think of this libary?
[](https://www.arduino.cc/en/ArduinoModbus/ArduinoModbus)
[](https://github.com/arduino-libraries/ArduinoModbus)

PR #53 (2022-12-19): Fixes and adds to modbus library and examples
FIX #36
- Some fixes to issues pointed by clients and detected in our internal tests.
- Implemented some missing functionalities like process of coil requests.
closed 2022-12-19 by pmmarquez