CONTROLLINO-PLC/CONTROLLINO_Library #37: Debugging Controllino

1simone0, 2020-03-03:
Hello everybody, I have 2 question:
The first, most important: How can I debug a Controllino? I couldn't find any information on the datasheet and user manual...
The second: is Controllino compatible with MPLAB X IDE since I can work on Atmel MCU's with it?

CONTROLLINO-Support, 2020-03-03:
Hello,
MCU in MINI does not have JTAG at all, so you can only use ISP interface for programming. All ISP pins are available at the pinheader. See [this application note for more details](http://ww1.microchip.com/downloads/en/appnotes/atmel-0943-in-system-programming_applicationnote_avr910.pdf). At MAXI and MEGA it is theoretically possible to access the JTAG, but it is necessary to dismount the device and do some modifications of the HW.

1simone0, 2020-03-05:
Thanks for your reply. I've never used the ISP interface but reading the page from your link it seems I can't debug but only program the Controllino using ISP, is it right? I will buy a Controllino Maxi automation and probably I will use Atmel Studio as IDE. Is in this case dismounting the Controllino the only way to debug? Will I need a debugger (like the Atmel ICE) or what else?
Any other solution?

BR
Simone

CONTROLLINO-Support, 2020-03-05:
Hi Simone, yes, the ISP interface is only for programming. You need to use JTAG port for programming. The JTAG pins are available at the pinheader, but due to protection circuitry it is not possible to use them through the CONTROLLINO pinheader for this purpose. For sure you will need Atmel ICE or PICkit 4 debugger.

1simone0, 2020-03-05:
Thanks, I got it! 
So, what kind of HW modification should I do for use JTAG port? Is there a guide?

catx23, 2020-03-06:
Hi,
I can't see my Serial.println messages ? 
thanks

1simone0, 2020-04-06:
Hello, I haven't had an answer yet... Can you repay please? I will start to work on it soon.

CONTROLLINO-Support, 2020-04-06:
Hello, I am sorry for the missing answer. We are preparing some tutorial for you. Stay tuned!

CONTROLLINO-Support, 2020-04-10:
Hello, I am sorry, but I have bad news for you.
During preparation of the tutorial we have found out that it is not possible to do these manual HW changes on a device which is already assembled. Without access to the bottom side of the middle board inside the CONTROLLINO you are not able to make it. So, the device has to be prepared for the JTAG debugging during production. Please [contact us](https://www.controllino.biz/contact/) if you need such a modified device. 
Thanks for understanding!

1simone0, 2020-11-22:
Hi, I was wondering if I can use MPLAB IDE to program (and debug) a Controllino MAXI and re-reading the first post of this issue I noticed that I have already asked but you probably missed that question. I take this opportunity to ask you also if there is any difference to use Atmel ICE or PICkit 4 debugger on Atmel Studio.
Same question on MPLAB in case Controllino is compatible withit.
Thank you!

BR,
Simone

CONTROLLINO-Support, 2020-11-22:
Hi Simone,

you are right, I have missed your question about MPLAB in the first post. I am sorry!

To be honest I have never tried to use MPLAB with Atmega MCU.

I always use
Atmel Studio + Atmel ICE + Atmega MCU
or
MPLAB + PICkit + PIC MCU  

But in general - if MPLAB and PICkit support AVR microcontrollers via ISP then it should work.

Best regards,
Lukas

Mr-Bart-Simpson, 2023-01-12:
For the sake of completeness (and as it may be helpfull for others): 

It is possible to debug Controllino "in software". I'm using **Visual Studio Code** with the **PlatformIO** extension for development.
**Be aware that this means, you cannot use `Serial` any longer in your code**

For debugging with that, you have to do the following steps:

1. Add a reference to the _avr-debugger_ library
2. Set `debug_tool = avr-stub` and  `debug_port = COMX` in plaformio.ini 
3. Add `#include "avr8-stub.h"` to your files
4. Add a call to `debug_init();` at the beginning of your `setup()` func
5. "Patch" the Controllino library. As using `Serial` conflicts with the debugging library, the `Controllino_PrintTimeAndDate` func has to be removed. 

I suggest that there should a) be a #define for the patch and b) an addition to the documentation in the future. This would make it easier for newcomers to get started with a _real_ debugger (and not doing Serial.print() as a by far inferior way to debug)

pmmarquez, 2025-02-18:
This issue will go to closed, the info is available there anyway.

closed 2025-02-18 by pmmarquez