CONTROLLINO-PLC/CONTROLLINO_Library #39: Controllino Maxi Automation 10V Analog Input

Trisiohno, 2020-09-22:
Hi Everyone,

I am using the Controllino Maxi Automation to receive a 0-10V analog Input and to send the corresponding 10 bit Value to another device.

If I set the Input Voltage to 0V I receive 0 and if I set it to 10V I receive 1023 so all is good at both ends.
Where I have the problem is all the Values in between, for some reason the analog input data is not converted to the correct 10 bit value e.g. with a input voltage of 0.5V I receive an outgoing value of 49 while I believe it should be 51.2 and with a input voltage of 8V I receive an outgoing value of 835 which should be 818.4.

Do you have any idea for why this is happening?
Is there anything specific I have to keep in mind while using the 10V analog Input? 

Tristan

simeon9696, 2021-04-15:
Hi Tristan, I had a similar issue. Is your device being powered with exactly 24.0V ?

sven1123, 2021-06-04:
In the newer post of Simeon I posted my theory of why this might happen.

pmmarquez, 2022-11-18:
Hello folks, this is how the 0-10V input is wired to microcontroller in the MAXI Automation hope this helpful to understand better how this measurements came to be: 
[image]

closed 2022-11-18 by pmmarquez