CONTROLLINO-PLC/CONTROLLINO_Library #40: Controllino IoT on AWS

1simone0, 2020-11-28:
Hi, I was wondering if it's possible to upload data on AWS IoT server using just Controllino via built in Ethernet port but I couldn't find any answer on the web and in Controllino documentations. Also tried to look for WIZnet 5100 and AWS compatibility but nothing.
How can I be sure that's possible to upload and receive datas from AWS IoT using a Controllino?
What should I check?
Anyone tried that?
I know also that AWS have some restrictions about security certificates if it can helps...
Any answer is welcome!

BR,
Simone

CONTROLLINO-Support, 2020-11-30:
Hi Simone, I have never tried it with AWS IoT, but in principle it should work - I assume that the MQTT protocol is the correct way. But in general such powerful IoT servers are more suitable for some more powerful devices (running Linux or at least some kind of a RTOS). I tried to connect to MS Azure one year ago and it was really painful! (so I cannot recommend it at all.) 
Maybe you can check [this article](https://aws.amazon.com/blogs/compute/building-an-aws-iot-core-device-using-aws-serverless-and-an-esp32/) as a starting point. 

Good luck!
Lukas

janibert, 2020-12-01:
Have a look at AWS GreenGrass.  My recommendation is to use some suitable device in your local network, like a NAS or a RasPi or some home-server that can run the GreenGrass core or a container with the core. Let the GreenGrass core do all the encrypted communication with AWS IoT over the internet. Inside your LAN (behind a firewall), GreenGrass can communicate with your CONTROLLINOs without encryption or other calculation intensive security. You can still use MQTT, or maybe CoAP, Modbus/TCP, or a self-invented lightweight TCP or UDP protocol.

closed 2022-11-18 by pmmarquez