CONTROLLINO-PLC/CONTROLLINO_Library #41: Strange Issue When Using AREF to measure 1.1V Bandgap

EmperorArthur, 2020-12-07:
Hello,

I am experiencing a strange issue when using the AREF pin (External Analog Reference) to measure the 1.1V internal reference bandgap voltage, and am curious if you could share how AREF is connected internally.

Looking at the [manual](https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega640-1280-1281-2560-2561-Datasheet-DS40002211A.pdf) on P.269, the ATMega chip has an internal 1.1V Bandgap reference.  However, the manual says it is only accurate to within +/-0.1V!

It should be possible to read that reference voltage like any other ADC pin, by merely setting the ADC's internal Mux.  That works, and using the normal AVcc reference voltage, everything works.  However, when using AREF, it mis-reads the bandgap as higher by almost 200mv!  What makes it strange is that effect happens even if Vcc is connected to AREF.

This may be an ATMega design flaw, but I am curious how AREF is handled internally.  Is there just a capacitor between it and ground, or is there anything else on the front end pin-header board that may be causing the issue?

It would be possible to check for front end effects by bridging AREF and AVcc together with the front end board removed.  However, I really do not want to risk doing something like that.

[Example Code](https://github.com/rlogiacco/VoltageReference/blob/master/VoltageReference.cpp) (from someone else) for measuring the bandgap using AVcc.

EmperorArthur, 2020-12-08:
Okay, I have done some digging and came across [this post](https://electronics.stackexchange.com/questions/30549/atmega328-external-aref-connection).  Based on it and the manual, there is a 32kΩ internal resister between the AREF pin and ground.  Most posts recommend a 5kΩ resistor on AREF so the AVCC can still be used without damaging the chip.  This would create a voltage divider, which could explain what I'm seeing.

However, if I set the ADC to use AREF, I can measure 32kΩ to ground with a multimeter!  Which points to the Controllino not having that resistor.  Plus I would need a 169kΩ to have the effect I am seeing.  Which seems to mean this is not the problem.

CONTROLLINO-Support, 2020-12-18:
Hello,
 
At CONTROLLINO MAXI and MEGA, AREF pin is conected from pinheader through ESD protection and 5k Ohm series resistor directly to the ATMEGA processor.
 
On CONTROLLINO MINI, AREF is accessible only inside the device and you have to use this 5k resistor externally.
 
BR
Jarda

EmperorArthur, 2021-01-20:
New year new findings. @CONTROLLINO-Support you may find this interesting.

I have taken the time to examine the AREF pin in detail, from the perspective of someone who does not have access to the schematic.  This was all done on a single Controllino Maxi Power.

# Physical Observations
I examined the pin with the housing on, and the Controllino running with `analogReference(EXTERNAL)`.  The AREF pin has an open circuit voltage of 3mV, and an immeasurably small.  When measuring the resistance with a good multi-meter, depending on the polarity of the leads, it measured either 32.54kΩ or 31.26Ω.  Regardless of the Voltage source, it was impossible to detect any current draw by the AREF pin.

It is not possible to measure the AREF pin when the Controllino is not running.

These results show no sign of the 5kΩ resistor.  The fact that there was no noticeable current draw is also puzzling, as the 32kΩ AVR resistor should have pulled at least a few micro Amps.  It may be measurement error, or there could be something like an Op-Amp present.

# Calculated Observations
In order to determine what the AREF was doing, the ADC had to measure a known value.  I chose to measure the 1.1V internal bandgap reference.  Without work, this is an inaccurate measurement, since the AVR documentation indicates it can vary by +/- 0.1V.  However, it is possible to read the exact voltage of the bandgap using the ADC with Vcc as the reference voltage.  Measuring Vcc with an accurate multi-meter then allowed me to determine the bandgap voltage relatively precisely.

Based on the 5kΩ resistor, I would have expected that the equation for the real reference voltage would be.
```
AREF_REAL = AREF * 32/(5+32)
```

Unfortunately, this does not match the expected results, and is off by quite a large margin.  Applying different voltages to the AREF pin I found the equation that best models the behavior is really:
```
AREF_REAL = AREF * 32/(6+32) + 0.003
```

The Controllino is acting as though there is a 6kΩ protection resistor, and the 3mV must be taken into consideration.

CONTROLLINO-Support, 2021-01-26:
Hello Arthur,
 
for sure there is assembled resistor 4k99 1% in the series between AREF CONTROLLINO pin on the pinheader and the AREF processor input (verified on the documentation and also physically on the board.
 
Resistance of the internal AREF pull down is specified in the processor datasheet without the limits as “typical value” so as you already mentioned it can be not exactly 32k. So I would say this is the difference you are measuring (also including the measurement tolerances).
 
Best Regards
Jarda

EmperorArthur, 2021-02-04:
Okay.  I am seeing a few different things.  Let me just spit out some numbers, and you might see why I have been having issues.  

I measured the internal bandgap voltage 3 different ways using the ADC.

| Measurement Mode | Measurement Voltage | ADC Result | Voltage Calculation Including ideal divider (if required) | Calculated Voltage |
|------------------|---------------------|------------|-----------------------------------------------------------|--------------------|
| DEFAULT          | 3.900               | 281/282    | Vref /1024 * ADC                                          | 1.070 / 1.074      |
| EXTERNAL         | 2.497               | 523/524    | (Vref * 32/(32+4.99)) /1024 * ADC                         | 1.103 / 1.105      |
| EXTERNAL         | 3.900               | 334        | Vref * (32/(32+4.99)) /1024 * ADC                         | 1.100              |

Meanwhile, direct measurement of the AREF pin with the reference voltage set to the 1.1V bandgap voltage gave me 1.083 volts.  As I mentioned previously, when measuring the pin with the mode set to EXTERNAL and nothing connected, I detected 3mv.

The truly puzzling part is the ~10mv difference between observed bandgap reference via multi-meter and calculated via AVCC.  Even then, the numbers really do not match up.

The unit is currently in the field and working with surprisingly good accuracy, merely by calibrating the voltage divider to that 1.07V reading, and applying the 3mV offset.

When the unit is back and I have time to work on it, I have another round of testing planned.  Specifically, I may hook up an external voltage source to an ADC pin on the header
* Measure some items using the 1.1V bandgap as the AREF
* Measure some items using Vcc on AREF
* Measure some items using 2.497V on AREF
* Measure some items using AVCC
* Perform ADC calibration as in the [AVR application note](http://ww1.microchip.com/downloads/en/appnotes/atmel-2559-characterization-and-calibration-of-the-adc-on-an-avr_applicationnote_avr120.pdf) for each of the four modes.

\*  That's on the bosses, and I'm working on convincing them to fix that.

closed 2022-11-18 by pmmarquez