CONTROLLINO-PLC/CONTROLLINO_Library #42: Interrupts from Ethernet

polymer-coder, 2021-03-17:
Hi! I have the controllino maxi automation and I was wondering if there's anyway to use generate an interrupt when the ethernet receives data? 

I did see [this article](https://garagebox.org/arduino/342-complete-soft-of-explaination-on-interrupt-imr-usage-of-ethernet-shield.html) for the WS100 chip but it requires soldering a jumper to the pin.

CONTROLLINO-Support, 2021-03-17:
Hi, yes it is. As you can see on the CONTROLLINO MAXI pinout the Wiznet interrupt pin is connected to the Atmega2560 pin PE6. [image]
 
This is not a standard Arduino pin, but we have defined it as a pin 71 in the CONTROLLINO library header file. 
[image]

PE6 is INT6 of the Atmega2560 microcontroller.
[image]

Good luck and share with us the results!
Lukas

polymer-coder, 2021-03-19:
Hi, 

I ended up using websockets into do bidirectional communication over local ethernet 

The server using node.js and [ws](https://www.npmjs.com/package/ws)

```
const WebSocket = require('ws');

const wss = new WebSocket.Server({ port: 3005 });

wss.on('connection', function connection(ws) {
  ws.on('message', function incoming(message) {
    console.log('received: %s', message);
  });

  
  setInterval(() => {
    ws.send(`${new Date()}`);
  }, 1000);
});
```

And on the Controllino the arduino websocket library. It is very important that you install the ATmega branch of this library *https://github.com/Links2004/arduinoWebSockets/tree/ATmega*:

```
#define MONITOR_SPEED               921600

#include <SPI.h>
#include <Ethernet.h>
#include <Arduino.h>
#include <Controllino.h>
#include <WebSocketsClient.h>

WebSocketsClient webSocket;

//MAC address for controllino - these can be manually assigned 
//but cannot conflict with any other address on the same network
byte mac[] = {0xAE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

// Set the static IP address to use if the DHCP fails to assign
static IPAddress ip(192, 168, 0, 177);
static IPAddress myDns(192, 168, 0, 1);
static EthernetClient client;

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

    switch(type) {
        case WStype_DISCONNECTED:
            Serial.println("[WSc] Disconnected!\n");
            break;
        case WStype_CONNECTED:
            {
                Serial.print("[WSc] Connected to url: ");
                Serial.println((char *)payload);
                // send message to server when Connected
                webSocket.sendTXT("Connected");
                
            }
            break;
        case WStype_TEXT:
            Serial.print("[WSc] get text: ");
            Serial.println((char *)payload);
            // send message to server
            // webSocket.sendTXT("message here");
            break;
        case WStype_BIN:
            Serial.print("[WSc] get binary length: ");
            Serial.println(length);
           // hexdump(payload, length);

            // send data to server
            // webSocket.sendBIN(payload, length);
            break;
    }

}

void setup()
{

    // Open serial communications and wait for port to open:
    Serial.begin(MONITOR_SPEED);  //921600
    while (!Serial)
    {
        ; // wait for serial port to connect. Needed for native USB port only
    }

    // start the Ethernet connection:
    Serial.println("Initialize Ethernet with DHCP:");
    if (Ethernet.begin(mac) == 0)
    {

        Serial.println("Failed to configure Ethernet using DHCP");
        
        // Check for Ethernet hardware present
        if (Ethernet.hardwareStatus() == EthernetNoHardware)
        {
            Serial.println("Ethernet shield was not found.");
            Serial.println("Program will continue to run without shield but will not report to web server");
        }

        if (Ethernet.linkStatus() == LinkOFF)
        {
            Serial.println("Ethernet cable is not connected.");
        }

        // try to congifure using IP address instead of DHCP:
        Ethernet.begin(mac, ip, myDns);
    }

    byte macBuffer[6];              // create a buffer to hold the MAC address
    Ethernet.MACAddress(macBuffer); // fill the buffer
    Serial.print(" MAC address : ");
    for (byte octet = 0; octet < 6; octet++)
    {
        Serial.print(macBuffer[octet], HEX);
        if (octet < 5)
        {
            Serial.print(':');
        }
    }
    Serial.println("");
    Serial.print("  DHCP assigned IP : ");
    Serial.println(Ethernet.localIP());

    webSocket.begin("192.168.0.189", 3005);
    webSocket.onEvent(webSocketEvent);
    
}

void loop()
{
   webSocket.loop();

  Ethernet.maintain();
  if (!client.connected()) {
    client.stop();
  }
  
}
```

This should print the current date and time on your serial monitor: 
```
[WSc] get text: Fri Mar 19 2021 10:13:12 GMT-0400 (Atlantic Standard Time)
[WSc] get text: Fri Mar 19 2021 10:13:13 GMT-0400 (Atlantic Standard Time)
[WSc] get text: Fri Mar 19 2021 10:13:14 GMT-0400 (Atlantic Standard Time)
[WSc] get text: Fri Mar 19 2021 10:13:15 GMT-0400 (Atlantic Standard Time)
[WSc] get text: Fri Mar 19 2021 10:13:16 GMT-0400 (Atlantic Standard Time)
```

And on your server: 
`received: Connected`

closed 2021-03-19 by polymer-coder