CONTROLLINO-PLC/CONTROLLINO_Library #43: ADC reading higher than it should be

simeon9696, 2021-04-15:
Hi, 

I'm using the analogRead function to just print the ADC representation of the signal. I know that the range of values @24V is 0-26.4V and for a safety reserve (which I don't completely understand) the ADC actually uses a reference(?) of 31V? 

So 0->31V is mapped onto 0->1023. That's understood. 

So I expected 24V to be (31/1023)/24 = 792 but instead its 825. Is my understanding wrong?

sven1123, 2021-06-04:
I don't know this for sure, but i think it's because they use the A/D converter of the microcontroller which is powered with 5V. When you use a 24V power supply and like you are saying it uses an 31V reference the scale means 31/6.14 = +- 5.05

So the Controllino divides your analog input voltage by 6,14 to use the internal 5V A/D converter. And because of all this conversions and scaling there might be a little deviation in the actual value that your Controllino reads.

I am not an expert, but I think my theory might be right.

pmmarquez, 2022-11-18:
Hello folks, the explanation of the scale factor is that there is a voltage divider, so any applied voltage on the input must be divided by the scale factor to determine the applied voltage on the microcontroller pin, so if you power the Controllino with 12V and you are applying 10V to an input: (10/3.06) ≈ 3.27V on the microcontroller pin.
[image]