CONTROLLINO-PLC/CONTROLLINO_Library #44: Wrong pins assignment in Controllino.h

Uskok, 2021-05-25:
Hi,

The pins are wrong assigned inside Controllino.h for MEGA

Wrong:
  #define CONTROLLINO_PIN_HEADER_DIGITAL_IN_16 **18**
  #define CONTROLLINO_PIN_HEADER_INT_00 18
  #define CONTROLLINO_SCREW_TERMINAL_DIGITAL_IN_16 **18**
  #define CONTROLLINO_SCREW_TERMINAL_INT_00 18
  
  #define CONTROLLINO_PIN_HEADER_DIGITAL_IN_17 **19**
  #define CONTROLLINO_PIN_HEADER_INT_01 19
  #define CONTROLLINO_SCREW_TERMINAL_DIGITAL_IN_17 **19**
  #define CONTROLLINO_SCREW_TERMINAL_INT_01 19
  
  **Correct:**
  #define CONTROLLINO_PIN_HEADER_DIGITAL_IN_16 38
  #define CONTROLLINO_SCREW_TERMINAL_DIGITAL_IN_16 38
   
  #define CONTROLLINO_PIN_HEADER_DIGITAL_IN_17 39
  #define CONTROLLINO_SCREW_TERMINAL_DIGITAL_IN_17 39
  
#define CONTROLLINO_PIN_HEADER_DIGITAL_IN_18 40
#define CONTROLLINO_SCREW_TERMINAL_DIGITAL_IN_18 40

 #define CONTROLLINO_PIN_HEADER_INT_00 18
 #define CONTROLLINO_SCREW_TERMINAL_INT_00 18

 #define CONTROLLINO_PIN_HEADER_INT_01 19
 #define CONTROLLINO_SCREW_TERMINAL_INT_01 19

CONTROLLINO-Support, 2021-05-31:
Hi,

yes, you are right! Thanks for the information. We will fix it soon.

Best regards,
Lukas

JasekJiri, 2021-06-02:
Hello

The Controllino.h is fixed.
Each new installation will include this fix, and when a new release is released, all libraries will be updated.

Best Regards
Jiri Jasek

closed 2021-06-02 by JasekJiri