CONTROLLINO-PLC/CONTROLLINO_Library #46: Support in IDE 2.0 and arduino-cl

gbataille, 2021-06-23:
Hi folks,

So I'm discovering the world of both arduino and PLC, so I'm not well equipped to figure things out on my own.

I have a controllino that work great with arduino IDE 1.8.x and with which we are controlling our machine.

Now, being a software engineer, I like 
* living on the edge, with the latest version (arduino IDE 2.0)
* using CLI tools (arduino-cli)

With both those tools, I'm able to use "regular" arduino board, but when using controllino, I get the same error in both environments:

`Compilation error: Error: 2 UNKNOWN: fork/exec /Users/gbataille/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++: bad CPU type in executable`

with `arduino-cli`, I can get a verbose output (that does not seem much helpful)

```
gbataille ~/Doc…DC/BiU_Corn/plc [1] $ arduino-cli compile --fqbn CONTROLLINO_Boards:avr:controllino_mega controllino --verbo
se
Using board 'controllino_mega' from platform in folder: /Users/gbataille/Library/Arduino15/packages/CONTROLLINO_Boards/hardware/avr/3.1.0
Using core 'arduino' from platform in folder: /Users/gbataille/Library/Arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/Users/gbataille/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Users/gbataille/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/Users/gbataille/Library/Arduino15/packages/CONTROLLINO_Boards/hardware/avr/3.1.0/variants/Controllino_mega /var/folders/cx/327j2f6j2j19z_tyjz5gqmtc0000gp/T/arduino-sketch-3F38DC76432B994275324F06CC5233C0/sketch/controllino.ino.cpp -o /dev/null

Error during build: fork/exec /Users/gbataille/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++: bad CPU type in executable
```

I have the intuition that it comes from the board definition file, but I would not know how that works.

Would you know?
Are those tools really not supported or is it just a matter of a small configuration?

Thanks

CONTROLLINO-Support, 2021-06-23:
Hi,

you are right. The tools are not supported yet. We have tried Arduino 2.0 some time ago and it was very unstable and almost useless. So we decided to wait until it will grow up from the "experimental state" into official release 01. The same for CLI - let´s wait until it is at least version 01. 

Nevertheless the build message is strange, because  controllino_mega and  -mmcu=atmega2560 is correct.

Please try to get some explanation from the 2.0/CLI developers, why the definition file (correct for old Arduino IDE and also for PlatformIO) does not work.

Good luck!
Lukas

gbataille, 2021-06-24:
Hey @CONTROLLINO-Support 

@perl1234 answered very quickly with this

> Note the avr-gcc version: 4.9.2-atmel3.5.3-arduino2. This is a very outdated version. Support for 32-bit applications was dropped from macOS in the 10.15 release. Arduino rebuild the macOS toolchains to 64-bit in response (arduino/Arduino#8974), but this version of avr-gcc dates from before that time.
> 
> So the Controllino boards platform maintainers will simply need to make a new release that uses the modern AVR toolchain. Information about that here:
> https://arduino.github.io/arduino-cli/dev/package_index_json-specification/#platforms-definitions

I hope this can help you speed up support for the latest platforms.

CONTROLLINO-Support, 2021-06-24:
OK, thanks, we will check this! Lukas

JasekJiri, 2021-06-28:
Hello

We tested the newer version of avr-gcc tools and we changed the "tools Dependencies" in HW package definition.
So compilation even in Arduino IDE 2.0 etc should be OK now.
See readme FAQ 14.

closed 2021-06-28 by JasekJiri