CONTROLLINO-PLC/CONTROLLINO_Library #48: vscode+PlatformIO does not flash when delay(variableName) is used. delay(100) works.

greenveg, 2021-09-22:
This code compiles but does not flash:
```
#include <Arduino.h>
#include <Controllino.h>  

long unsigned int valveCycleTime = 20;

void setup() {  
  pinMode(CONTROLLINO_D0, OUTPUT);
  
  digitalWrite(CONTROLLINO_D0, HIGH);   
  delay(valveCycleTime);                          
  digitalWrite(CONTROLLINO_D0, LOW);    
  delay(100);   
}

void loop() {}
```

When changed to `delay(100)` it compiles and flashes no problem.

Using `int` or `uint32_t` does not seem to matter.

CONTROLLINO-Support, 2021-09-22:
Hello, I have tried the same code with Arduino IDE and CONTROLLINO MAXI and it works fine.

Which variant of CONTROLLINO do you have?

Thanks,
Lukas

greenveg, 2021-09-22:
Older MINI. 
Will try with a freshly bought MAXI tomorrow!

pmmarquez, 2022-11-18:
This is definitely working without issue

closed 2022-11-18 by pmmarquez