CONTROLLINO-PLC/CONTROLLINO_Library #49: Controllino boards not found

edmsNG, 2021-12-01:
Hello,
I have an issue with instaling Controllino boards on Arduino IDE. I did all steps that are required (install Controllino library, paste URL in Preferences), but when I open Board Manager and type in search CONTROLLINO nothing shows up. Does anyone have any idea why and how can I fix this?

[image]

CONTROLLINO-Support, 2021-12-01:
Hello, please double check the URL link. Maybe there are some whitespaces there?

It works fine at my side

[image]

edmsNG, 2021-12-01:
Hello,
Thank you for quick response. I duble checked. I pasted in this: https://raw.githubusercontent.com/CONTROLLINO-PLC/CONTROLLINO_Library/master/Boards/package_ControllinoHardware_index.json
and still no luck. Is this URL corect? 
Are there some aditional settings that I need to change?

[image]

CONTROLLINO-Support, 2021-12-01:
No, it should work Out-of-the-box and yes, the URL is correct.

closed 2022-11-18 by pmmarquez