CONTROLLINO-PLC/CONTROLLINO_Library #52: Unclear if RTC alarm functionality is in 3.07 or not

greenveg, 2022-05-24:
Both [this example](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/Common/DemoRTCAlarm/DemoRTCAlarm/DemoRTCAlarm.ino) and the instructions [here](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library#rtc-functions) tells me that there is RTC alarm functionality implemented.

I have the latest version, 3.0.7 installed in PlatformIO / vsCode and it just gives me a standard _'Controllino_ClearAlarm' was not declared in this scope_ when I try to compile code with `Controllino_ClearAlarm()` or `Controllino_SetAlarm(int, int)`.

Downloaded the 3.0.7 source and had a look, no alarms found there either.

Am I blind/stupid or is it simply something for 3.0.8?

Thanks!

CONTROLLINO-Support, 2022-05-24:
Hi greenveg!

You are right. The Alarm stuff is not in the 3.0.7. Try to manually replace the  Controllino.h and Controllino.cpp files by the lates ones from the github.

And we will release the 3.0.8 in the meantime ...

Best regards,
Lukas

greenveg, 2022-05-24:
Another note:
In your[ RTC example](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/Common/DemoRTCAlarm/DemoRTCAlarm/DemoRTCAlarm.ino) you have a slight error on row 76 I think:

`if (digitalRead(CONTROLLINO_RTC_INTERRUPT) = 1)`

Pretty sure it should be `== 0` since you also have it on a pullup in the same example:
`pinMode(CONTROLLINO_RTC_INTERRUPT, INPUT_PULLUP);`

greenveg, 2022-05-24:
Alarm and timer functionality would be super helpful!

Ping @Columbo818, looks like it you who perhaps know why ClearAlarm() is not working?

Columbo818, 2022-05-24:
Good catch @greenveg - I'm at work at the moment but will sort a bug fix once I'm home and can test.

@CONTROLLINO-Support I implemented this because you hadnt. I was hoping you'd take my code and move forwards with it, but from what I can see you've just left it as-is, and havent released any updates since. 

Your software is lacking core features and demos, and your definition of "open source" seems to be "contact us and we'll decide if you're worthy" - Its very disappointing, and is why I dont use a Controllino anymore.

greenveg, 2022-05-30:
Bump @Columbo818 🙂

I tried having a look at what datasheet for the RTC module says about setting and clearing alarms to see if I could sort it out but these waters are too deep for me I'm afraid 😅

Mr-Bart-Simpson, 2023-01-12:
As the discussion here goes a bit into bugfixing the alarm funcs in the lib, I'd like to add another comment here.

First of all, I have to mention, that I'm not a experienced C/C++ developer (my background is more on C#...) and therefore, this may or may not be a real issue!

When directly including the source into my project (using PlatformIO in Visual Studio Code), I get a warning on the `Controllino_SetAlarm()` func. 

```
char Controllino_SetAlarm(unsigned char *aHour, unsigned char *aMinute) {	
	unsigned char TimeDate [2]={aMinute,aHour};
...
```
The warning states, that the assignment of the TimeDate array is a problem, because _a value of type "unsigned char *" cannot be used to initialize an entity of type "unsigned char"_

greenveg, 2023-02-06:
Bump!
Is this fixed yet? Any timeline for a 3.0.8 release? 
I'm debating if I should buy more Controllinos or try one of your [competitors](https://industruino.com/shop/product/industruino-ind-i-o-d21g-2?category=2) (mostly for fun, I've used Controllinos since 2017 and still love them)

CONTROLLINO-Support, 2023-03-10:
I am sorry for the delayed 3.0.8 Release. We were doing some personal changes here. It should be done in the next two weeks.

Columbo818, 2023-06-06:
@greenveg https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/pull/59

pmmarquez, 2024-02-07:
Is #59 fixing the issue @greenveg and @CONTROLLINO-Support ?

PR #66 (2024-03-28): Adding RTC alarm changes
Fix #52 by including changes on #59
closed 2024-03-28 by CONTROLLINO-Support