CONTROLLINO-PLC/CONTROLLINO_Library #57: Question: Monitoring the 24V

AceTecNic, 2023-03-10:
Is it possible to monitor the 24V voltage? I connected the controller to USB and when I restart my machine, the controller keeps running because of the USB supply?
Thanks!

CONTROLLINO-Support, 2023-03-10:
Hi,

Yes, correct, the MCU and Ethernet is running also from the USB supply.

It is not possible to monitor 24V internally, but you can externally connect it to one of the analog inputs. 

Lukas

AceTecNic, 2023-03-10:
Over which analog input does this work?
I have from the power supply directly to CONTORLLINO_SCREW_TERMINAL_ANALOG_ADC_IN_10 (int spannungsVersorgung) and query this with "if (spannungsVersorgung == 0){motorenAus();}".
I think the input can't be read when the voltage drops? Or am I wrong there?

CONTROLLINO-Support, 2023-03-10:
I have used CONTROLLINO MEGA, screw terminal analog input A10 and this sketch

```
#include <SPI.h>
#include <Controllino.h>

int sensorValue = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("A10 reading:");
  
}

void loop() {
  // put your main code here, to run repeatedly:
  sensorValue = analogRead( CONTROLLINO_A10 );
  Serial.println(sensorValue);
  delay(1000);
}

```

And I can see following output when switch ON a OFF the 24V power:

A10 reading:
796
796
796
796
796
796
796
795
795
4
0
0
0
798
798
798
798
45
0
0
0
798
798
798

AceTecNic, 2023-03-10:
Ahhh perfect, thanks!
Now it works.

Problem solved.
Thank you for this great product :)

CONTROLLINO-Support, 2023-03-10:
You are welcome :)