CONTROLLINO-PLC/CONTROLLINO_Library #60: Failed uploading: cannot execute upload tool

bitmarte, 2023-10-10:
Hi @CONTROLLINO-Support,
sketch upload doesn't work on Arduino IDE 2.2.1 because of an error like this:

`Failed uploading: cannot execute upload tool: fork/exec .... : bad CPU type in executable`

There is also a discussion on your forum ([here](https://community.controllino.com/forums/discussion/after-update-to-arduino-ide-2-0-1-uload-is-no-longer-possible/)).
I think there is a problem with _avrdude_ lib version.

I tried on two MacOSX laptop, intel and arm, with the same issue.

andytheengineerguy, 2023-11-09:
Hi @bitmarte, I'm experiencing this same issue with a Controllino MAXI right now. Did you find a solution?

SLE-LL-95, 2023-11-10:
Hello @andytheengineerguy!
I also had the same issue recently.
My solution was to either compile for Arduino Mega (instead of selecting Controllino) or use Arduino IDE 1.x.
It is a pity that Controllino Team is not responding...

bitmarte, 2023-11-10:
> Hi @bitmarte, I'm experiencing this same issue with a Controllino MAXI right now. Did you find a solution?

Hi @andytheengineerguy,
the problem is inside the **CONTROLLINO Library** which is not really mantained from the creator.
You can find below the current options/workaround:

1. workaround: force **avrdude/6.3.0-arduino17** usage from the IDE
3. workaround: compile and upload with an old Arduino IDE version (eg. 1.8.x)
4. fix: update **CONTROLLINO Library** which is opensource library but we have to spend a lot of time in order to understand whole process because we are not the creator

@CONTROLLINO-Support known the issue and they promised the fix [here ](https://community.controllino.com/forums/discussion/after-update-to-arduino-ide-2-0-1-uload-is-no-longer-possible/) but no news, bad news...

Controllino Boards are a great project without support, totaly; we have to take it in our mind when we plan to use this gorgeous project in our production env, real world PLC and automations.

pmmarquez, 2024-01-31:
Is already in the pipeline to update to the latest version of tools and packages used by Arduino AVR board support packages which fix this issues like @bitmarte says force avrdude/6.3.0-arduino17 is a provisional solution [here](https://github.com/arduino/arduino-cli/issues/1336#issuecomment-1900247915) is how.
Sorry for the delay.

PR #63 (2024-02-29): Tools version updated
- Fix #60 
- Update tools to the latest version like in Arduino AVR Boards
- Adding Arduino OTA tool
- CONTROLLINO Boards version to 3.1.1
closed 2024-02-29 by CONTROLLINO-Support