CONTROLLINO-PLC/CONTROLLINO_Library #76: Ethernet.linkStatus() is not supported on Controllino Mega

Spike433, 2025-02-17:
In your example [ModbusTCPServer.ino](https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/blob/master/examples/Expand_Modbus/TCP/ControllinoServer/ModbusTCPServer.ino),  checking is ethernet cable connected can't work. I'm referring to this 
```c++
if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");
  }
```

I was wondering why it isn't working until I found out that in [Ethernet.h](https://github.com/arduino-libraries/Ethernet/blob/master/src/Ethernet.h) , 

```c++
enum EthernetLinkStatus {
	Unknown,
	LinkON,
	LinkOFF
};

enum EthernetHardwareStatus {
	EthernetNoHardware,
	EthernetW5100,
	EthernetW5200,
	EthernetW5500
};
```

### Documentation

Tells you whether the link is active. LinkOFF could indicate the Ethernet cable is unplugged or defective. This feature is only available when using the W5200 and W5500 Ethernet controller chips.

### Hardware status of controllino mega

Code returns 
```c++
  Serial.println(Ethernet.hardwareStatus()); --> `Serial Console = 1 which is W5100 that is not supported for link status`

enum {
EthernetNoHardware
EthernetW5100
EthernetW5200
EthernetW5500
}
```

## Example for testing

```c++
/*
  Link Status

  This sketch prints the Ethernet link status. When the
  Ethernet cable is connected the link status should go to "ON".
  NOTE: Only WIZnet W5200 and W5500 are capable of reporting
  the link status. W5100 will report "Unknown".
  Hardware:
   - Ethernet shield or equivalent board/shield with WIZnet W5200/W5500
  Written by Cristian Maglie
  This example is public domain.
*/

#include <SPI.h>
#include <Ethernet.h>

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  //Ethernet.init(10);  // Most Arduino shields
  //Ethernet.init(5);   // MKR ETH Shield
  //Ethernet.init(0);   // Teensy 2.0
  //Ethernet.init(20);  // Teensy++ 2.0
  //Ethernet.init(15);  // ESP8266 with Adafruit FeatherWing Ethernet
  //Ethernet.init(33);  // ESP32 with Adafruit FeatherWing Ethernet

  Serial.begin(9600);
}

void loop() {
  auto link = Ethernet.linkStatus();
  Serial.print("Link status: ");
  switch (link) {
    case Unknown:
      Serial.println("Unknown");
      break;
    case LinkON:
      Serial.println("ON");
      break;
    case LinkOFF:
      Serial.println("OFF");
      break;
  }
  delay(1000);
}
```

## Question 
Do you have other elegant solution that will reset client connection after cable is unplugged, because modbus tcp server thinks client is still connect connected and when you plug ethernet cable back, client is trying to reconnect while server didn't end connection gracefully.

pmmarquez, 2025-02-18:
This is not an issue inherent to the library.
@Spike433 please contact us to techsupport@controllino.com we will gladly try to help you find a solution there.

closed 2025-02-18 by pmmarquez

Spike433, 2025-02-23:
Hello @pmmarquez,  techsupport is not responding, are they on vacation?

pmmarquez, 2025-03-03:
> Hello [@pmmarquez](https://github.com/pmmarquez), techsupport is not responding, are they on vacation?

Hi @Spike433 I am sure you will get response soon anyway i will take a look to your request